Livox-SDK/livox_ros_driver2 #1: Problem with roslaunch_rviz

Minsu-Kim-KNU, 2022-08-24:
Hi,

I'm triying to use HAP with 20.04 and ROS noetic.

I follow all the steps in this repository, but I can't make the run my HAP.
In Windows10 I check my HAP was working well.

Therefore, I have two questions.

1. How to run my HAP like in the windows?
2. What is the bd in the launch file?

Thanks.

[image]

QiXueZhong, 2022-11-17:
change the IP 192.168.1.5 in 'config_ROS1/HAP_config.json' to you computer IP

Livox-Infra, 2023-01-16:
Hi @Minsu-Kim-KNU ,
1. Pls check your HAP whether is in the same LAN with your PC and whether the IPs and Ports are set properly in the config_ROS1/HAP_config.json.
2. bd is a deprecated field for HAP. No need to concern about it.