Livox-SDK/livox_ros_driver2 #7: 我不能source这个目录

Kt-Janice, 2022-12-13:
lhg@KT:~/livox_ros_driver2$ source devel/setup.bash
bash: devel/setup.bash: 没有那个文件或目录

Livox-Infra, 2022-12-13:
1. livox_ros_driver2 文件夹应放置在某个工作目录的 src/ 路径下，比如路径结构是 ~/ws_livox/src/livox_ros_driver2 (而非直接放在 ~/ 路径下）
2. 然后参考 [README](https://github.com/Livox-SDK/livox_ros_driver2/blob/1.0.0/README.md) 的 2.1 / 2.3 / 2.4 节的操作步骤，应该可以解决你遇到的问题

Kt-Janice, 2022-12-13:
[image]
还有一个问题，我json文件的IP地址已经设置成电脑IP，但还是无法驱动雷达

Livox-Infra, 2022-12-13:
抱歉，这个 sample 需要 sudo 权限执行；
稍晚些会优化一下

pjnhlh, 2022-12-13:
这个问题 fixed 了，请 git pull 一下 Livox SDK2 的代码

Livox-Infra, 2022-12-13:
这个问题 fixed 了，请 git pull 一下最新的 Livox SDK2 代码

Kt-Janice, 2022-12-13:
我刚试过，用最新的driver2就可以了，谢谢你

closed 2022-12-14 by Livox-Infra