Livox-SDK/livox_ros_driver2 #8: When Livox Avia support?

FPSychotic, 2022-12-16:
When Livox Avia support?

Thanks

Livox-Infra, 2022-12-18:
hi @FPSychotic , 
Livox ROS Driver2 is not available for Livox Avia now and future...

[Livox ROS Driver](https://github.com/Livox-SDK/livox_ros_driver) fits for it, or is there any problem when using Livox ROS Driver?

FPSychotic, 2022-12-18:
Basically I want use the repetitive mode of my Livox Avia From 6 months ago with ROS, and I would like use it with ROS2 too. Does allow the ROS1 driver both things?
I asked before a few times but no answer.

Livox-Infra, 2022-12-19:
hi @FPSychotic , you can try [livox viewer 0.11.0](https://terra-1-g.djicdn.com/65c028cd298f4669a7f0e40e50ba1131/Download/update/Livox%20Viewer%200.11.0.zip) to setup repetitive mode, then use [livox ros2 driver](https://github.com/Livox-SDK/livox_ros2_driver) with ROS2.

FPSychotic, 2022-12-19:
Ohh, thanks , I didn't know that could be done. I will report back if it works ,maybe helps others

closed 2023-04-03 by FPSychotic