Livox-SDK/livox_ros_driver2 #9: convert lvx2 to rosbag

ZhenpengChenCode, 2022-12-21:
您好，请问，什么时候支持将lvx2的点云数据转成ros包？

Livox-Infra, 2023-01-09:
后续会支持，可能会以一个单独小工具的形式提供转换功能，目前未完全确定方案

sukhrajklair, 2023-01-16:
I see there is sample data on Livox Mid360 product page but its in lvx2 format. Is it possible for you to upload it in rosbag or pcd(s) format? I tried the converter tool in LivoxViewer2 but it creates one pcd file for all of the scans.

arvrschool, 2023-03-02:
I have the problem too.

cartooh, 2023-03-10:
It's not working, what's wrong?

```cpp:lds_lvx.cpp
//
// The MIT License (MIT)
//
// Copyright (c) 2022 Livox. All rights reserved.
//
// Permission is hereby granted, free of charge, to any person obtaining a copy
// of this software and associated documentation files (the "Software"), to deal
// in the Software without restriction, including without limitation the rights
// to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
// copies of the Software, and to permit persons to whom the Software is
// furnished to do so, subject to the following conditions:
//
// The above copyright notice and this permission notice shall be included in
// all copies or substantial portions of the Software.
//
// THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
// IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
// FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
// AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
// LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
// OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
// SOFTWARE.
//

#include <stdio.h>
#include <unistd.h>
#include <string.h>
#include <functional>
#include <memory>
#include <thread>
#include <condition_variable>
#include <mutex>

#include "lds_lvx.h"
#include "lvx_file.h"
#include "comm/pub_handler.h"

#include <unistd.h>
#include <sys/param.h>

namespace livox_ros {

// std::condition_variable LdsLvx::cv_;
// std::mutex LdsLvx::mtx_;
std::atomic_bool LdsLvx::is_file_end_(false);

LdsLvx::LdsLvx(double publish_freq) : Lds(publish_freq, kSourceLvxFile), is_initialized_(false) {
}

LdsLvx::~LdsLvx() {
}

int LdsLvx::Init(const char *lvx_path) {
  if (is_initialized_) {
    printf("Livox file data source is already inited!\n");
    return false;
  }

#ifdef BUILDING_ROS2
  DisableLivoxSdkConsoleLogger();
#endif

  printf("Lds lvx init lvx_path:%s.\n", lvx_path);
  inf_lvx2_.open(lvx_path, std::ios_base::binary);
  if (!inf_lvx2_)
  {
    char path[MAXPATHLEN];
[... 692 lines cut ...]
    {

    }

    BasePackHeader header;
    std::vector<ExtendRowPoint> raw_point;
};

class BaseHalfPackDetail
{
public:
    BaseHalfPackDetail()
    {

    }
    ~BaseHalfPackDetail()
    {

    }

    BasePackHeader header;
    std::vector<ExtendHalfRowPoint> raw_point;
};
#pragma pack(pop)
```

NHMMing, 2023-05-05:
> It's not working, what's wrong?
> 
> * lds_lvx.cpp
> 
> ```c++
> //
> // The MIT License (MIT)
> //
> // Copyright (c) 2022 Livox. All rights reserved.
> //
> // Permission is hereby granted, free of charge, to any person obtaining a copy
> // of this software and associated documentation files (the "Software"), to deal
> // in the Software without restriction, including without limitation the rights
> // to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
> // copies of the Software, and to permit persons to whom the Software is
> // furnished to do so, subject to the following conditions:
> //
> // The above copyright notice and this permission notice shall be included in
> // all copies or substantial portions of the Software.
> //
> // THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
> // IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
> // FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
> // AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
> // LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
> // OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
> // SOFTWARE.
> //
> 
> #include <stdio.h>
> #include <unistd.h>
> #include <string.h>
> #include <functional>
> #include <memory>
> #include <thread>
> #include <condition_variable>
> #include <mutex>
> 
> #include "lds_lvx.h"
> #include "lvx_file.h"
> #include "comm/pub_handler.h"
> 
> #include <unistd.h>
> #include <sys/param.h>
> 
> namespace livox_ros {
> 
> // std::condition_variable LdsLvx::cv_;
> // std::mutex LdsLvx::mtx_;
> std::atomic_bool LdsLvx::is_file_end_(false);
> 
> LdsLvx::LdsLvx(double publish_freq) : Lds(publish_freq, kSourceLvxFile), is_initialized_(false) {
> }
> 
> LdsLvx::~LdsLvx() {
> }
> 
> int LdsLvx::Init(const char *lvx_path) {
>   if (is_initialized_) {
>     printf("Livox file data source is already inited!\n");
>     return false;
>   }
> 
> #ifdef BUILDING_ROS2
>   DisableLivoxSdkConsoleLogger();
> #endif
> 
>   printf("Lds lvx init lvx_path:%s.\n", lvx_path);
>   inf_lvx2_.open(lvx_path, std::ios_base::binary);
>   if (!inf_lvx2_)
[... 702 lines cut ...]
>     }
> 
>     BasePackHeader header;
>     std::vector<ExtendRowPoint> raw_point;
> };
> 
> class BaseHalfPackDetail
> {
> public:
>     BaseHalfPackDetail()
>     {
> 
>     }
>     ~BaseHalfPackDetail()
>     {
> 
>     }
> 
>     BasePackHeader header;
>     std::vector<ExtendHalfRowPoint> raw_point;
> };
> #pragma pack(pop)
> ```

it maybe has some difference in file header between lvx and lvx2

sssolitude, 2024-01-25:
一年过去了，有办法了嘛

Lixc23tsinghua, 2024-03-04:
笑死了，我也遇到这个问题了。。。。mid360不配用目标检测

DHA-Tappuri, 2025-01-30:
I had the same problem too and created a simple Python script.
You can find it here: [lvx2_to_rosbag](https://github.com/DHA-Tappuri/lvx2_to_rosbag)
It seems to work fine with the indoor sample data: [Indoor Sample Data](https://terra-1-g.djicdn.com/65c028cd298f4669a7f0e40e50ba1131/Mid360/Indoor_sampledata.lvx2)

championAI, 2025-07-03:
两年过去了，有办法了嘛

FelixCooper1026, 2025-07-03:
> 两年过去了，有办法了嘛

可以参考一下我的项目 https://github.com/FelixCooper1026/lvx2_to_rosbag

championAI, 2025-07-04:
> > 两年过去了，有办法了嘛
> 
> 可以参考一下我的项目 https://github.com/FelixCooper1026/lvx2_to_rosbag

可以使用，感谢！！！