Livox-SDK/livox_ros_driver2 #10: can't see polintcloud in rviz

ZhenpengChenCode, 2022-12-22:
Hello, I run the command "roslaunch livox_ros_driver2 rviz_HAP.launch", it shows that lidar has been successfully initialized, but I can't see any point cloud on rviz.
[image]

ahmadchalhoub, 2023-01-09:
Hello. I am running into the same problem @ZhenpengChenCode. Have you found a solution for this yet?

ZhenpengChenCode, 2023-01-12:
> Hello. I am running into the same problem @ZhenpengChenCode. Have you found a solution for this yet?

No yet.

ZhenpengChenCode, 2023-01-12:
> 

Hi, I found out that the problem was that our ethernet cable was connected to the wrong port. You can check if you can ping 192.168.1.100. If the pings are different, it means that the data link is not connected.

Livox-Infra, 2023-01-13:
> > 
> 
> Hi, I found out that the problem was that our ethernet cable was connected to the wrong port. You can check if you can ping 192.168.1.100. If the pings are different, it means that the data link is not connected.

hi @ZhenpengChenCode , sorry for late reply. Is there any problem for connection to the lidar?

ZhenpengChenCode, 2023-01-15:
> hi @ZhenpengChenCode , sorry for late reply. Is there any problem for connection to the lidar?

We found that the point cloud data could not be received because of the wrong connection of the Ethernet port. It's resolved now, thanks.

closed 2023-01-16 by Livox-Infra