Livox-SDK/livox_ros_driver2 #12: Could not run ros2 launch rviz_HAP.py

WinRobotics, 2023-01-14:
Hi,
I face an issue where I could not load the SDK library for livox_ros_driver2_node when I tried to run ros2 launch rviz_HAP.py. I have a screenshot of the issue I face below.
[image]
May I ask if anyone faces the same issue, and if there is any solution? Thank you!

danicannt, 2023-01-26:
I have the exact same problem, either in ROS Foxy and ROS Galactic

danicannt, 2023-01-26:
I have fixed using sudo ldconfig after installing the SDK, ldconfig creates the necessary links and cache to the latest shared libraries in the directories specified on the command line, in the file /etc/ld.so.conf,  and in trusted directories  (/lib and /usr/lib). I think it shall be mentioned anywhere in the documentation.

andy-zhuo-02, 2023-05-18:
> 

yeap. I am disturbed by this annoying problem too. Thanks for your help. I highly recommend that this issue should be alarmed in readme guidance.