Livox-SDK/livox_ros_driver2 #15: humble run launch can't publish point  cloud only imu

Michael-ztk, 2023-01-29:
ros2 launch livox_ros_driver2 rviz_MID360_launch.py 
[INFO] [launch]: All log files can be found below /home/tangkyle/.ros/log/2023-01-29-17-38-06-617720-tangkyle-zhou-55675
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [livox_ros_driver2_node-1]: process started with pid [55676]
[INFO] [rviz2-2]: process started with pid [55678]
[livox_ros_driver2_node-1] [INFO] [1674985086.708310342] [livox_lidar_publisher]: Livox Ros Driver2 Version: 1.0.0
[livox_ros_driver2_node-1] [INFO] [1674985086.708567266] [livox_lidar_publisher]: Data Source is raw lidar.
[livox_ros_driver2_node-1] [INFO] [1674985086.708580068] [livox_lidar_publisher]: Config file : /home/tangkyle/ws_livox/install/livox_ros_driver2/share/livox_ros_driver2/launch_ROS2/../config/MID360_config.json
[livox_ros_driver2_node-1] LdsLidar *GetInstance
[livox_ros_driver2_node-1] config lidar type: 8
[livox_ros_driver2_node-1] successfully parse base config, counts: 1
[livox_ros_driver2_node-1] [INFO] [1674985086.710287874] [livox_lidar_publisher]: Init lds lidar success!
[livox_ros_driver2_node-1] GetFreeIndex key:livox_lidar_201435328.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, lidar type:8, handle:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[... 353 lines cut ...]
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, can not get index, lidar type:8, device_num:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] Storage point data failed, lidar type:8, handle:2986453184.
[livox_ros_driver2_node-1] Can not get index, the livox lidar type:8, handle:2986453184
[livox_ros_driver2_node-1] found lidar not defined in the user-defined config, ip: 192.168.1.178
[livox_ros_driver2_node-1] GetFreeIndex key:livox_lidar_2986453184.
[livox_ros_driver2_node-1] begin to change work mode to 'Normal', handle: 2986453184
[livox_ros_driver2_node-1] successfully change work mode, handle: 2986453184
[livox_ros_driver2_node-1] successfully enable Livox Lidar imu, ip: 192.168.1.178
[livox_ros_driver2_node-1] [INFO] [1674985089.710465210] [livox_lidar_publisher]: livox/imu publish use imu format

topic list only have imu data
I check the code ,found key not match, but don't know why

Hezip, 2023-01-30:
Same problem for me~

Hezip, 2023-01-30:
> 

I have solved my problem, So I need to change the Lidar IP(192.168.1.12) to the correct address(192.168.1.1xx). XX is the last 2 numbers of SN.   
Hope this information could help you!

Michael-ztk, 2023-01-30:
> 

first thx,
but my default lidar IP is 192.168.1.178(78 is the last 2 numbers of SN),and the last 2 logs show the IP, so I don't know how to do

Michael-ztk, 2023-01-30:
> > 
> 
> I have solved my problem. I change the Lidar IP(192.168.1.12) to the correct address(192.168.1.1xx). XX is the last 2 numbers of SN. Hope this information could help you!

thank you！！！ i finally know how to fix

closed 2023-01-31 by Michael-ztk

pbszcc, 2024-05-24:
May I ask if you have resolved it using this method