Livox-SDK/livox_ros_driver2 #16: Multiple NIC

Griwa88, 2023-01-31:
Is there a way to have two host IP in config, if two NIC's are used for two separate HAPs?

NIC Port 1
Host 192.168.1.10 to HAP 192.168.1.11
NIC Port 2
Host 192.168.2.10 to HAP 192.168.2.11

I would like to synchronize clocks without buying expensive gPTP capable switches, and I have two ports capable of 802.1AS so this would be the most economical/practical route.

Griwa88, 2023-01-31:
Nevermind, I figured it out.
Made 2 launch files, changed node name in second launch file, and launched two nodes with separate configs.

closed 2023-01-31 by Griwa88

0NNS0, 2023-05-31:
Hi,I have encountered the same problem as you. Can you provide me with your two launch and config files for reference?
Thanks!