Livox-SDK/livox_ros_driver2 #23: MID360:The point cloud is not displayed

finedcl, 2023-02-26:
I have connected the radar mid360 in the virtual machine, run rviz_mid360.launch, the initialization is successful, but there is no point cloud displayed in the rviz interface, frame is livox_frame,
 the warning message is: No tf data. Actual error: Fixed Frame [livox_frame] does not exist.
and  run rostopic echo /livox/lidar，data is shown.

arvrschool, 2023-03-06:
I have the same problem in ubuntu..

finedcl, 2023-03-06:
> 
do not use the virtual machine

naveenkumar2208, 2023-03-16:
@finedcl Did it solve for you?

naveenkumar2208, 2023-03-16:
@arvrschool Did it solve for you?

finedcl, 2023-03-17:
@naveenkumar2208  yes,standalone ubuntu systems can display point clouds

FPSychotic, 2023-04-03:
in standalone windows needed deactivate firewall, in ubuntu standalone in x86 and Arm64 will show the pointclouds, nut livox viewer2 wont work in arm64. in the ros config file there is ports , I would check that ports are available in your system and ip address

LeonardoGuevara, 2023-07-08:
I have a similar problem, I also get the warning message is: No tf data. Actual error: Fixed Frame [livox_frame] does not exist.
But in my case when I run rostopic echo /livox/lidar, no message is published. I'm running in in Ubuntu 20.04. Any solutions?

oscarpoudel, 2023-09-15:
I also have a similar problem as Leonardo . I tried installing it on both jetson nano as well as intel cpu ubuntu 20.04 but the point cloud is still not showing in Rviz.

FPSychotic, 2023-09-15:
> I also have a similar problem as Leonardo . I tried installing it on both jetson nano as well as intel cpu ubuntu 20.04 but the point cloud is still not showing in Rviz.
I assume you got it work in livox SDK , but you cannot in ROS. The best way to test the lidar i running Fast-lio 2 with the mid-360 specific launch. If you run for the driver , be sure you use the rviz launch. If still don't show points, be sure you set global_frame as " livox_frame", mayne you need write it at hand  and be sure the topic is pointcloud2 format and called /libox/lidar.
For sure the drivers works. But also you need setup properly the IP address in the lidar nodel

oscarpoudel, 2023-09-18:
I checked it with Fast Lio too its the same. The cloud points are not showing. I have set the ip in the config file to that of my system and lidar ip to as specified in the Manual.
The issue is shown in the figure.

[image]

[image]

kosmonauta144, 2023-09-21:
Hi Guys, 

I had similar problem and here is the fix which helped me.

Livox lidar seems to have IP adress isue, it does not have static adress or it's not configured with the .json file.

I connected the livox mid360 lidar to mikrotik router or router with ip scanner. I scanned subnetwork and get this adresses

[image]

192.168.1.5 is my computer
192.168.1.144 is second device connected which is Livox

so I changed IP adress of the lidar in the .json file to 192.168.1.144 and it works

[image]

no TF data still remains, but the point cloud is received. I hope it helps you.

I though that json file is made to set IP adress for the lidar, but it has to be set to static through router or switch to work. I hope it helps.

georgflick, 2023-09-21:
Somewhere in the docs I found out, that the initial IP address is set to `192.168.1.100 + last two digits of serial number`

> Hi Guys,
>
> I had similar problem and here is the fix which helped me.
>
> Livox lidar seems to have IP adress isue, it does not have static adress or it's not configured with the .json file.
>
> I [...]

FPSychotic, 2023-09-21:
It is in every guide ,you need put the last two numbers of the lidar s/n

Prabuddhi-05, 2023-10-19:
> I have connected the radar mid360 in the virtual machine, run rviz_mid360.launch, the initialization is successful, but there is no point cloud displayed in the rviz interface, frame is livox_frame, [...]

I'm encountering this issue in Ubuntu 22.04 while working with ROS 2 Humble. However, when I run 'ros2 topic echo /livox/lidar,' no data is displayed. Is there a solution for this?

Prabuddhi-05, 2023-10-23:
> I'm encountering this issue in Ubuntu 22.04 while working with ROS 2 Humble. However, when I run 'ros2 topic echo /livox/lidar,' no data is displayed. Is there a solution for this?

The issue was resolved by turning off the firewall to ensure that some of the ports were no longer blocked.

dpbnasika, 2024-01-07:
did the no TF data issue fixed or someone can able to fix ?, i now try to add the tf and it does not display the axis and imu axis also it is not recognized as topic in rviz topics tab, but i can able to see the imu topic in terminal. Any help is appreciated.

Prabuddhi-05, 2024-01-08:
Dear sir,

As of now, I haven't attempted to solve this specific problem. I will let you know if I come across any potential solutions.

Thanks and best regards,
Prabuddhi

xincheng-yang, 2024-06-25:
I just checked the mid-360 instructions and found out that the lidar ip address should be set to 192.168.1.1xx, which xx is last two digits of your lidar's serial number. 
my pc ip address: 192.168.1.5 
subnet mask: 255.255.255.0

Prabuddhi-05, 2024-06-25:
[like]  Prabuddhi Wariyapperuma (26619... reacted to your message:

adijams01, 2025-06-21:
I was using it on ROS2 Humble via docker from raspi 5 (ubuntu 24.04), I was getting pointcloud data before but no tf data on rviz, i simply added a static tf between livox_frame and base_link and error was gone