Livox-SDK/livox_ros_driver2 #24: Init lds lidar successfully. but no subsequent data.

LouxiDahu, 2023-02-26:
[ INFO] [1677403048.659921645]: Livox Ros Driver2 Version: 1.0.0
data source:0.
[ INFO] [1677403048.661836128]: Data Source is raw lidar.
[ INFO] [1677403048.662091837]: Config file : /home/rm/rm_sentry_ws/src/livox_ros_driver2/config/MID360_config.json
LdsLidar *GetInstance
config lidar type: 8
successfully parse base config, counts: 1
[2023-02-26 17:17:28.662] [console] [info] Livox lidar logger disable.  [parse_cfg_file.cpp] [Parse] [84]
[2023-02-26 17:17:28.662] [console] [info] Data Handler Init Succ.  [data_handler.cpp] [Init] [49]
[2023-02-26 17:17:28.662] [console] [info] Init livox lidars succ.  [device_manager.cpp] [Init] [153]
GetFreeIndex key:livox_lidar_2214701248.
[ INFO] [1677403048.662475619]: Init lds lidar successfully!

qlibp, 2023-02-28:
same problem encounter

LouxiDahu, 2023-02-28:
> same problem encounter

Use 12V power supply and wait for a while.

CQ-Quest, 2023-05-04:
> > same problem encounter
> 
> Use 12V power supply and wait for a while.

same problem encounter. Use 12V power but nothing changed. What should I do ？

LouxiDahu, 2023-05-04:
> 

remain
1.using the 12v dc power
2.check the network interface and sure it working well.
3.trying to remove the crystal head and link it again.

closed 2023-05-04 by LouxiDahu