Livox-SDK/livox_ros_driver2 #26: Why not use two branches to separate `ros1` and `ros2 `

Ericsii, 2023-03-01:
Why don't you open two branches to separately support ROS1 and ROS2? In order to directly compile the driver use `catkin build` or `colcon build` like any other ROS repos.

Livox-Infra, 2023-03-13:
We do not separate ROS1 and ROS2 Driver because most of the code are the same. 
Soon we would provide a new 'build script', which enable user to use `catkin build` or `colcon build` like other ROS repos (but still not to separate branches)

rdu-weston, 2023-11-15:
> We do not separate ROS1 and ROS2 Driver because most of the code are the same. Soon we would provide a new 'build script', which enable user to use `catkin build` or `colcon build` like other ROS repos (but still not to separate branches)

Honestly, providing a build script is not a very great idea. Yes, it simplifies the building by not asking users to manually do the steps in the script, but on the other hand you're forcing users to follow a non-ROS-conventional way to handle packages. It's fine if you only keep the Livox packages in this workspace but not so convenient if you also have other packages, which I believe is the more common case. Would you consider accepting #46 and merging it as soon as possible?

closed 2024-10-23 by Ericsii

Chick92, 2025-05-01:
Not having separate ROS1 and ROS2 branches is an awful idea. It causes so many problems with both ROS1 and ROS 2 in docker.