Livox-SDK/livox_ros_driver2 #31: Is it possible to run livox_ros_driver2 in ubutu Docker on window host?

Squek, 2023-03-16:
Hi sir,
Is it possible to run livox_ros_driver2 in Ubuntu Docker on window host?
I have test livox_ros_driver2 work in Ubuntu docker container on Ubuntu host, 
docker command is docker run -it --name temp --network host containername:tag /bin/bash. 

But same docker container can not run normally on window host, 
Init lds lidar successfully. but no any subsequent data. message show as below:

LdsLidar *GetInstance
config lidar type: 8
successfully parse base config, counts: 1
[2023-03-16 07:43:35.298] [console] [info] Livox lidar logger disable. [parse_cfg_file.cpp] [Parse] [84]
[2023-03-16 07:43:35.298] [console] [info] Data Handler Init Succ. [data_handler.cpp] [Init] [49]
[2023-03-16 07:43:35.299] [console] [info] Init livox lidars succ. [device_manager.cpp] [Init] [153]
GetFreeIndex key:livox_lidar_1677830336.
[ INFO] [1678952615.300051300]: Init lds lidar successfully!

zvweaf, 2023-07-29:
Same problem, have you solved it?

UpppD7, 2024-01-22:
check your docker configuration:
           --net=host \
           --ipc=host \
it may work.