Livox-SDK/livox_ros_driver2 #34: Multiple lidars in different frames

halejo-as, 2023-03-30:
Hello,

Im trying to use to lidars 360 in different frames, if i run the driver only for one lidar works well but when two drivers are executing only one lidar is published, i already changed the multi_topic parameter to 1.  

could be a good idea has a example of launch multiple lidars in different topics and frames

KimHyung, 2023-03-31:
1.IP settings
I used the Livoxviewer  to set each lidar to have a different IP.
- lidar1 : HAP, 192.168.1.100
- lidar2 : HAP, 192.168.1.101
- HostPC : 192.168.1.5

2.change config file
{
  "lidar_summary_info" : {
    "lidar_type": 8
  },
  "HAP": {
    "lidar_net_info" : {
      "cmd_data_port": 56000,
      "push_msg_port": 0,
      "point_data_port": 57000,
      "imu_data_port": 58000,
      "log_data_port": 59000
    },
    "host_net_info" : {
      "cmd_data_ip" : "192.168.1.5",
      "cmd_data_port": 56000,
      "push_msg_ip": "",
      "push_msg_port": 0,
      "point_data_ip": "192.168.1.5",
      "point_data_port": 57000,
      "imu_data_ip" : "192.168.1.5",
      "imu_data_port": 58000,
      "log_data_ip" : "",
      "log_data_port": 59000
    }
  },
  "HAP": {
    "lidar_net_info" : {
      "cmd_data_port": 56000,
      "push_msg_port": 0,
      "point_data_port": 57000,
      "imu_data_port": 58000,
      "log_data_port": 59000
    },
    "host_net_info" : {
      "cmd_data_ip" : "192.168.1.5",
      "cmd_data_port": 56000,
      "push_msg_ip": "",
      "push_msg_port": 0,
      "point_data_ip": "192.168.1.5",
      "point_data_port": 57000,
      "imu_data_ip" : "192.168.1.5",
      "imu_data_port": 58000,
      "log_data_ip" : "",
      "log_data_port": 59000
    }
  },
  "lidar_configs" : [
    {
      "ip" : "192.168.1.100",
      "pcl_data_type" : 1,
      "pattern_mode" : 0,
      "extrinsic_parameter" : {
        "roll": 0.0,
        "pitch": 0.0,
        "yaw": 0.0,
        "x": 0,
        "y": 0,
        "z": 0
      }
    },
    {
      "ip" : "192.168.1.101",
      "pcl_data_type" : 1,
      "pattern_mode" : 0,
      "extrinsic_parameter" : {
        "roll": 0.0,
        "pitch": 0.0,
        "yaw": 0.0,
        "x": 0,
        "y": 0,
        "z": 0
      }
    }
  ]
}

3. changed the multi_topic parameter "1"

JanRBoth, 2023-04-05:
I have the same problem with two HAP's.

I also start two nodes for the respective HAPs. Multi_topic parameter is set to 1, IP's are different, Config file was adjusted as described here.

However, it seems that the first starting node goes through the config file and publishes both listed there.

In fact, two topics or pointclouds are published for me as well. But both with the same frame_id, which is not wanted although in the start launch file different frame_id parameters are passed.

In the driver code it seems that the frame_id is set as parameter but for all created publishers, so the parameter is always overwritten. Actually the option multi_topic should distinguish between different frame_id's, like it is done with the names of the topics.

If someone has a solution or I am wrong I would be very grateful.

zymouse, 2023-11-06:
https://github.com/Livox-SDK/livox_ros_driver2/pull/85
Problem solved, test passed

xiaoxueshengyao, 2023-11-07:
> #85 Problem solved, test passed

Thanks for your work, it works for me.

TZECHIN6, 2024-07-24:
First of all, I am agreed to have this feature to allow user to define each lidar publish frame, and the workaround in above works (hopefully a PR or official implementation will be deployed).

Second, I use another way to solve this issue by launching two different set of config.json and two nodes. Make sure the `host_net_info` ip address must be set, otherwise the config will be overwritten to all lidar due to boardcast.

TZECHIN6, 2024-07-30:
> First of all, I am agreed to have this feature to allow user to define each lidar publish frame, and the workaround in above works (hopefully a PR or official implementation will be deployed). [...]

Further tested, this approach is not stable. sometime the hostnet config will ignore the according lidar ip and set all the connected lidar to publish in a same port.

JoLichtenfeld, 2025-01-14:
Hi @TZECHIN6 , I am having the same problem. When running two MID-360 (in separate drivers, publishing on separate ros topics, same host_net_info IPs, different ports) only one of them actually sends data. It is always the one started later.
As you described, they both send to the same port (tested with wireshark), no matter what their config says.

Have you found a solution to this?

TZECHIN6, 2025-01-16:
Finally I have used above temp fix to let the driver to publish in different frames

JoLichtenfeld, 2025-01-16:
But as you wrote: 

> sometime the hostnet config will ignore the according lidar ip and set all the connected lidar to publish in a same port

Did you ever solve this?

TZECHIN6, 2025-01-16:
> But as you wrote: 
> 
> > sometime the hostnet config will ignore the according lidar ip and set all the connected lidar to publish in a same port
> 
> Did you ever solve this?

I mean I stick with the official setup for multi lidar configuration, in this way, you should not to specify port for each, to split the lidar point cloud, use above method to assign different frame to each lidar.

Hotsburp, 2025-02-20:
Hi, could i check if the temp fix that you all tried above can be done in ROS1?