Livox-SDK/livox_ros_driver2 #35: Jetson Orin NX ARM64  20.04 MID-360 error

FPSychotic, 2023-04-01:
`
nrover@nrover:~/LivoxViewer2 for Ubuntu v2.3.0$ ros2 launch livox_ros_driver2 msg_MID360_launch.py 
[INFO] [launch]: All log files can be found below /home/nrover/.ros/log/2023-04-01-09-48-14-095897-nrover-3980
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [livox_ros_driver2_node-1]: process started with pid [3982]
[livox_ros_driver2_node-1] [INFO] [1680338894.306635909] [livox_lidar_publisher]: Livox Ros Driver2 Version: 1.0.0
[livox_ros_driver2_node-1] [INFO] [1680338894.307152810] [livox_lidar_publisher]: Data Source is raw lidar.
[livox_ros_driver2_node-1] [INFO] [1680338894.307185643] [livox_lidar_publisher]: Config file : /home/nrover/livox2_ws/install/livox_ros_driver2/share/livox_ros_driver2/launch_ROS2/../config/MID360_config.json
[livox_ros_driver2_node-1] LdsLidar *GetInstance
[livox_ros_driver2_node-1] config lidar type: 8
[livox_ros_driver2_node-1] successfully parse base config, counts: 1
[livox_ros_driver2_node-1] bind failed
[livox_ros_driver2_node-1] Failed to init livox lidar sdk.
[livox_ros_driver2_node-1] [ERROR] [1680338894.311483220] [livox_lidar_publisher]: Init lds lidar fail!
`
My config file, same 192.168.1.50 in host and 192.168.1.110 as lidar will work in windows.
`
  "lidar_summary_info" : {
    "lidar_type": 8
  },
  "MID360": {
    "lidar_net_info" : {
      "cmd_data_port": 56100,
      "push_msg_port": 56200,
      "point_data_port": 56300,
      "imu_data_port": 56400,
      "log_data_port": 56500
    },
    "host_net_info" : {
      "cmd_data_ip" : "192.168.1.50",
      "cmd_data_port": 56101,
      "push_msg_ip": "192.168.1.50",
      "push_msg_port": 56201,
      "point_data_ip": "192.168.1.50",
      "point_data_port": 56301,
      "imu_data_ip" : "192.168.1.50",
      "imu_data_port": 56401,
      "log_data_ip" : "",
      "log_data_port": 56501
    }
  },
  "lidar_configs" : [
    {
      "ip" : "192.168.1.110",
      "pcl_data_type" : 1,
      "pattern_mode" : 0,
      "extrinsic_parameter" : {
        "roll": 0.0,
        "pitch": 0.0,
        "yaw": 0.0,
        "x": 0,
        "y": 0,
        "z": 0
      }
    }
  ]
}

`
maybe I dont know setup the Ethernet, first time I do in linux
[image]
[image]

FPSychotic, 2023-04-01:
my error, forgot files in ros2 must be modified in install folder instead src.

closed 2023-04-01 by FPSychotic

Hezhexi2002, 2023-06-02:
> my error, forgot files in ros2 must be modified in install folder instead src.

@FPSychotic Hi,I also meet the same problem on nuc11,could you please explain what does it mean "files in ros2 must be modified in install folder instead src",I have source the setup.sh in install folder but it didn't work at all

FPSychotic, 2023-06-03:
> @FPSychotic Hi,I also meet the same problem on nuc11,could you please explain what does it mean "files in ros2 must be modified in install folder instead src",I have source the setup.sh in install [...]

The launch files and config files to setup IP,  port ,launch that are needed to modify to adjust them to your PC are duplicated in SRC and INSTALL folders after compile, but are the ones in the Install folder the ones you need modify.