Livox-SDK/livox_ros_driver2 #36: Is repeating scanning mode supported for mid360?

leewo0, 2023-04-03:
The output point cloud makes no change after changing pattern mode to 1.

FPSychotic, 2023-04-03:
i dont know if that option is for that, but if you change it in the livox viewer2 will work later in ros, the mode will be active until you change it again. obviously would be better a option as the one you said and it works, but it is a 
workaround

leewo0, 2023-04-04:
@FPSychotic Thanks for your reply. I checked the lidar settings in the livox viewer2. The `Scan Pattern` had been changed to `Repetitive`(changing the option worked in the ros driver), but the output point cloud is still non repetitive in both livox viewer2 and ros driver. I noticed the firmware version is `13.17.01.11` and the product info is `DevType:Mid-360 FmT` in the setting menu. Is this feature not supported by my mid360?

FPSychotic, 2023-04-04:
I will check again and check fw version, but quite sure is working

FPSychotic, 2023-04-04:
> @FPSychotic <https://github.com/FPSychotic> That's a little werid. Today,
> I contacted DJI's after-sales staff(my mid360 was bought there), and he
> learned from the developer of Livox that mid360 does not currently support
> repeat scanning mode.

Need check but I read about it has one LoL

leewo0, 2023-04-26:
not supported yet

closed 2023-04-26 by leewo0

LotfiZ, 2024-02-06:
Hi Livox team, 

Are you planning to support Repetitive pattern for the mid 360 ? If yes, can we have a timeline estimation if its possible ? 

Thank you