Livox-SDK/livox_ros_driver2 #43: multi lidar time synchronization

h-gokul, 2023-04-19:
I am able to use this driver for a single lidar setup and publish point clouds. However, to increase the FOV, I have two hap lidars fit next to each other. They arent extrinsically calibrated yet. I want the data to be published in a single topic, therefore I will set the multi_topic flag accordingly. I would like to know if this driver can time synchronize the pointclouds by default when setting up two lidars.

This is the config file 
```
{
"lidar_summary_info" : {
	"lidar_type": 8
	},
"HAP": {
	"lidar_net_info" : {
		"cmd_data_port": 56000,
		"push_msg_port": 0,
		"point_data_port": 57000,
		"imu_data_port": 58000,
		"log_data_port": 59000
		},
	"host_net_info" : {
		"cmd_data_ip" : "192.168.1.5",
		"cmd_data_port": 56000,
		"push_msg_ip": "",
		"push_msg_port": 0,
		"point_data_ip": "192.168.1.5",
		"point_data_port": 57000,
		"imu_data_ip" : "192.168.1.5",
		"imu_data_port": 58000,
		"log_data_ip" : "",
		"log_data_port": 59000
		}
},
"HAP": {
	"lidar_net_info" : {
		"cmd_data_port": 56000,
		"push_msg_port": 0,
		"point_data_port": 57000,
		"imu_data_port": 58000,
		"log_data_port": 59000
		},
	"host_net_info" : {
		"cmd_data_ip" : "192.168.1.5",
		"cmd_data_port": 56000,
		"push_msg_ip": "",
		"push_msg_port": 0,
		"point_data_ip": "192.168.1.5",
		"point_data_port": 57000,
		"imu_data_ip" : "192.168.1.5",
		"imu_data_port": 58000,
		"log_data_ip" : "",
		"log_data_port": 59000
		}
},
"lidar_configs" : [
	{
	"ip" : "192.168.1.100",
	"pcl_data_type" : 1,
	"pattern_mode" : 0,
	"extrinsic_parameter" : {
		"roll": 0.0,
		"pitch": 0.0,
		"yaw": 0.0,
		"x": 0,
		"y": 0,
		"z": 0
		}
	},
	{
	"ip" : "192.168.1.101",
	"pcl_data_type" : 1,
	"pattern_mode" : 0,
	"extrinsic_parameter" : {
		"roll": 0.0,
		"pitch": 0.0,
		"yaw": 0.0,
		"x": 0,
		"y": 0,
		"z": 0
		}
	}
]
}

```

KimHyung, 2023-04-20:
I used messge_filter to sync time when i merged point_cloud.

you can try [this pointcloud_merger node](https://github.com/KimHyung/pointcloud_merger) to merged pointclouds with time_sync.

h-gokul, 2023-07-08:
I would like to obtain capture time synchronization between both lidar devices than just stitching the point clouds.

Mengfan-Source, 2024-09-25:
Hello, have you received the answer? Do you need extra time to align? I encountered the same problem.

Kham96, 2025-03-25:
Hello, have you solved the problem of syncing multiple sensors?

Mengfan-Source, 2025-03-25:
I haven't solved this problem yet. If possible, I hope to get your support.

---- Replied Message ----
| From | ***@***.***> |
| Date | 03/25/2025 16:44 |
| To | ***@***.***> |
| Cc | Xu ***@***.***>***@***.***> |
| Subject | Re: [Livox-SDK/livox_ros_driver2] multi lidar time synchronization (Issue #43) |

Hello, have you solved the problem of syncing multiple sensors?

—

Hello, have you solved the problem of syncing multiple sensors?

lexuantung9999, 2025-04-02:
Hello @Mengfan-Source I have a question. When you use 2 livox lidars with these setups: multi_topic = 0, publish_freq=10, which is the real frequency of topic /livox/lidar? 20Hz or 10Hz. The config multi_topic = 0 does not mean that all the points are merged in one topic, that mean the lidars publish independently in same topic, right?

Mengfan-Source, 2025-04-02:
In my experiment, I set different topic names for the two lidars through configuration files. Therefore, the data of the two lidars will be published on two separate topics, with each topic having a data publishing frequency of 10 Hz.

---- Replied Message ----
| From | ***@***.***> |
| Date | 04/02/2025 19:32 |
| To | ***@***.***> |
| Cc | Xu ***@***.***>***@***.***> |
| Subject | Re: [Livox-SDK/livox_ros_driver2] multi lidar time synchronization (Issue #43) |

Hello @Mengfan-Source I have a question. When you use 2 livox lidars with these setups: multi_topic = 0, publish_freq=10, which is the real frequency of topic /livox/lidar? 20Hz or 10Hz. The config multi_topic = 0 does not mean that all the points are merged in one topic, that mean the lidars publish independently in same topic, right?

—

Hello @Mengfan-Source I have a question. When you use 2 livox lidars with these setups: multi_topic = 0, publish_freq=10, which is the real frequency of topic /livox/lidar? 20Hz or 10Hz. The config multi_topic = 0 does not mean that all the points are merged in one topic, that mean the lidars publish independently in same topic, right?

lexuantung9999, 2025-04-03:
@Mengfan-Source do you have lidars there? can you try to set them in same topic, and tell me the frequency when using this option?