Livox-SDK/livox_ros_driver2 #45: Init lds lidar successfully. but no subsequent data. I need help!

CQ-Quest, 2023-05-04:
[image]
Init lds lidar successfully. but no subsequent data. I need help!
Ubuntu18.04  melodic
Livox HAP
changed IP and can be pinged normally.
[image]
use 12V power and no problem with network connection. But no data. 
What should I do?

hwan0806, 2023-05-25:
Same problem in mid360 case. Did you solved it? I'd appreciate if you can share any solutions!

CQ-Quest, 2023-05-26:
> 

remain
1.using the 12v dc power
2.check the network interface and sure it working well.
3.trying to remove the crystal head and link it again.check by ping lidar ip itself.(ping 192.168.1.100)

My problem is the step3 and sloved.

closed 2023-05-26 by CQ-Quest