Livox-SDK/livox_ros_driver2 #55: What is the "detected points property: Other", bits 4-5 in the tag field represent?

arihantlunawat, 2023-07-05:
Hi,

I see that a significant portion of the points (between 1% - 3% per scan) are setting the tag field bits 4-5 which is specified as "detected points property: Other" in the manual. Can I get more insight into this property?

arihantlunawat, 2023-07-05:
For quick reference:

[image]

```
full cloud size:      39936
low_conf_cloud size:  38763
not_low_conf_cloud size:  1173
Tag bins: 
[0 = 0 | 0 | 0]: 38763   [1 = 1 | 0 | 0]:   119   [2 = 2 | 0 | 0]:    86   [4 = 0 | 1 | 0]:   346   [5 = 1 | 1 | 0]:     2   [6 = 2 | 1 | 0]:     1   [16 = 0 | 0 | 1]:   618   [32 = 0 | 0 | 2]:     1
```

georgflick, 2023-07-06:
I had the same question and when I looked in the point cloud, the "other" points were always points near the lidar origin.

In the livox_viewer, when you select the noise filter, it also describes the "Other" points as "Filter dragging noises near the lidar".
[image]