Livox-SDK/livox_ros_driver2 #60: Access to Livox-SDK2

emuemuJP, 2023-07-14:
Currently, I can't access Livox-SDK2 repository.
https://github.com/Livox-SDK/Livox-SDK2/

Is this only my problem?

HopeCollector, 2023-07-16:
same problem

Snowbound9173, 2023-07-16:
Yeah someone removed the repository, it's still in google results and the livox website here: https://www.livoxtech.com/downloads 

But 404 and there's no reference anymore.

So you might wonder if someone had the bright idea to merge both SDKs to make 1 master to rule them all? But no. Even running minor edits to update livox_ros_driver2 CMakeLists.txt file to points at the SDK1 by updating: 

`  ## make sure the livox_lidar_sdk_static library is installed
  find_library(LIVOX_LIDAR_SDK_LIBRARY  liblivox_sdk_static.a    /usr/local/lib)
`

used to be: 
`
  ## make sure the livox_lidar_sdk_static library is installed
  find_library(LIVOX_LIDAR_SDK_LIBRARY  liblivox_lidar_sdk_static.a    /usr/local/lib)
`
But this still won't compile, to nobodies surprise.

purple-mesh, 2023-07-17:
@Livox-SDK any suggestion on what is the best work around for now?

atinfinity, 2023-07-19:
I noticed that the Livox-SDK2 has been re-released.
<https://github.com/Livox-SDK/Livox-SDK2/>

emuemuJP, 2023-07-19:
@atinfinity Thank you for telling us!
The issue has been resolved, so I am closing this issue. Thank you!