Livox-SDK/livox_ros_driver2 #66: launch problem terminate called after throwing an instance of 'std::bad_alloc'   what():  std::bad_alloc

kxc12689, 2023-08-08:
roslaunch livox_ros_driver2 msg_MID360.launch

Then initiating shutdown.

Here are the output:

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
================================================================================REQUIRED process [livox_lidar_publisher2-2] has died!
process has died [pid 7268, exit code -6, cmd /home/k/Project/ws_livox/devel/lib/livox_ros_driver2/livox_ros_driver2_node 47MDL610010087 __name:=livox_lidar_publisher2 __log:=/home/k/.ros/log/50db33b6-35c4-11ee-91d5-08bfb82db560/livox_lidar_publisher2-2.log].
log file: /home/k/.ros/log/50db33b6-35c4-11ee-91d5-08bfb82db560/livox_lidar_publisher2-2*.log
Initiating shutdown!
================================================================================

kxc12689, 2023-08-09:
> I have the same problem but during launching rviz_MID.launch but I don't know how to fix it. Does the error occur also with this launch?

All the launch files in livox_ros_driver2 about ROS1 have the problem. I think this function package have some bug undetected.

Snowbound9173, 2023-08-11:
Same error. This is my first time installing the driver after they pulled down the repository a few weeks ago and then only recently re-uploaded it. 

Were there changes?

trishmapow, 2023-08-14:
Same issue after "upgrading" to the re-released SDK2.

trishmapow, 2023-08-15:
Turned out to not be a problem with SDK. Reverted driver to v1.1.3 and now working again.
`git checkout 4f40ab789d7305207a3b371f635b114068b68601`

Ywz19960711, 2023-08-16:
I have not encountered this problem, are you using version 1.2.2  ? by the way,  The ros driver needs to be placed in the src directory.

kxc12689, 2023-08-16:
> I have not encountered this problem, are you using version 1.2.2 ? by the way, The ros driver needs to be placed in the src directory.

Yes, its v1.2.2. The package correctly places in workspace/src/.

Call-Me-Jesse, 2023-08-16:
> > I have not encountered this problem, are you using version 1.2.2 ? by the way, The ros driver needs to be placed in the src directory.
> 
> Yes, its v1.2.2. The package correctly places in workspace/src/.

I am using 1.2.1 and have not encountered this problem, you can try this version, maybe it will solve the problem

YCJin9, 2023-08-17:
I use LivoxViewer2 set host ip can solve this problem

Marabir, 2023-08-18:
> I use LivoxViewer2 set host ip can solve this problem

so I have to set host ip with LivoxViewer2? I will check it on Monday.

Call-Me-Jesse, 2023-08-19:
> > I use LivoxViewer2 set host ip can solve this problem
> 
> so I have to set host ip with LivoxViewer2? I will check it on Monday.

I found that 1.2.3 version does not have this problem, instead of set host ip with LivoxViewer2, try 1.2.3 version.

YCJin9, 2023-08-19:
> > > I use LivoxViewer2 set host ip can solve this problem
> > 
> > 
> > so I have to set host ip with LivoxViewer2? I will check it on Monday.
> 
> I found that 1.2.3 version does not have this problem, instead of set host ip with LivoxViewer2, try 1.2.3 version.

ok, i‘ll try it