Livox-SDK/livox_ros_driver2 #70: How to config two network interfaces

YCJin9, 2023-08-21:
I have two MID360 devices connected to two network interfaces, but it seems that the driver can only use one host IP. How should I configure this?

georgflick, 2023-08-21:
maybe start the driver two times with different configs? one for each interface?

YCJin9, 2023-08-21:
> maybe start the driver two times with different configs? one for each interface?

when I run second driver， the first driver‘s pointcloud will stop，and second driver output "update lidar failed status:-4". when I run any launch separately, they are ok. I try to run double launch in different docker container，ros only have the second lidar's topic.

apeoej333, 2023-08-22:
me too.

rviz_mixed launch file doesn't work.

rviz_MID360_A, rviz_MID360_B doesn't work(different host's port number or ip whatever...)

YCJin9, 2023-08-22:
I created a bridge and added two network interfaces, then I can settwo lidar in the same segment.

apeoej333, 2023-08-28:
> > maybe start the driver two times with different configs? one for each interface?
> 
> when I run second driver， the first driver‘s pointcloud will stop，and second driver output "update lidar failed status:-4". when I run any launch separately, they are ok. I try to run two launch files in different docker container，ros only have the second lidar's topic.

Hello, YCJin

Can you elaborate a bit more?

Please send it to [dahoon@maxst.com](mailto:dahoon@maxst.com) and thank you.

Call-Me-Jesse, 2023-09-05:
I read the source code of the parse_cfg_file.cpp file in SDK2, and I think this config file should solve your problem.
`{
  "MID360": {
    "lidar_net_info" : {
      "cmd_data_port"  : 56100,
      "push_msg_port"  : 56200,
      "point_data_port": 56300,
      "imu_data_port"  : 56400,
      "log_data_port"  : 56500
    },
    "host_net_info" : [
      {
	    "lidar_ip"       : ["192.168.1.10"],
        "host_ip"        : "192.168.1.5",
        "multicast_ip"   : "224.1.1.5",
        "cmd_data_port"  : 56101,
        "push_msg_port"  : 56201,
        "point_data_port": 56301,
        "imu_data_port"  : 56401,
        "log_data_port"  : 56501
      },
	  {
	    "lidar_ip"       : ["192.168.1.11"],
        "host_ip"        : "192.168.1.6",
        "multicast_ip"   : "224.1.1.5",
        "cmd_data_port"  : 56101,
        "push_msg_port"  : 56201,
        "point_data_port": 56301,
        "imu_data_port"  : 56401,
        "log_data_port"  : 56501
      }
    ]
  }
}
`

RahulRewale, 2025-04-25:
@YCJin9 
Were you able to figure this out? I have the same question.