Livox-SDK/livox_ros_driver2 #74: livox_ros_driver2 bind failed

h0uzd, 2023-08-29:
I downloaded Livox-SDK 2 and livox_ros_driver 2. when running logger from Livox-SDK 2, it works. But when trying to launch msg_MID360_launch.py and rviz_MID360_launch.py an error occurs. I'm using ROS 2 humble.output at startup msg_MID360_launch.py:
[image]
MID360_config.json:
[image]

Call-Me-Jesse, 2023-09-04:
The "ip" parameter in lidar_config must be equal to the ip of the lidar?

Yuxuan1024, 2023-11-25:
I face the same issue. Did you solve this problem?

bionik-berlin, 2023-12-07:
I have the same problem. The ip parameter is correct. Did you get a result?

FightInYoung, 2023-12-20:
you need to check the path of the MID360_config.json. which can be changed in the launch file. In your picture, the  path of the MID360_config is "/home/mdsp/disk/ws_livox/install/livox_ros_driver2/share/livox_ros_driver2/launch_ROS2/../config/MID360_config.json". You can check the ip

proxi666, 2024-02-20:
did anyone resolve this error?

Victor-Boyd, 2024-04-02:
Make sure you double check all ip addresses, both for the lidar and for the host machine, missing one digit will ruin the whole things so it's very important this double check.

SasaKuruppuarachchi, 2024-10-29:
Do a ```colcon build``` after updating the config file this will copy the configs to the install folder

Eku127, 2025-06-12:
> Do a `colcon build` after updating the config file this will copy the configs to the install folder

That helps, thanks!