Livox-SDK/livox_ros_driver2 #75: What does `line` (0~3) in Livox Mid360's pointcloud2(PointXYZRTLT) format mean?

LarryDong, 2023-08-30:
Hi,

For mid360, the `xfer_format` can be 0, which is "Livox pointcloud2(PointXYZRTLT) pointcloud format".
I printed the `line` for `L`, and see the value is 0,1,2,3.

What does line mean? The Readme says:
> uint8   line                 # **laser** number in lidar

Is **laser** the **ring** number for each scan? Are there only 4 lines (rings) at each scanning?

LarryDong, 2023-09-04:
> Yes, mid360 is a four-wire scan.

Thanks Jesse~

af-doom, 2024-04-02:
> 是的，mid360 是四线扫描。

您好，我采集的数据里tag是0是正常的吗

af-doom, 2024-04-02:
> 您好，我采集的数据里tag是0是正常的吗

> Hi,
>
> For mid360, the `xfer_format` can be 0, which is "Livox pointcloud2(PointXYZRTLT) pointcloud format". I printed the `line` for `L`, and see the value is 0,1,2,3.
>
> What does line mean? The [...]

您好，我采集的数据里tag始终为0，line为0-3是正常的吗，另外我在这个仓库下https://github.com/hku-mars/LiDAR_IMU_Init下载的mid360数据line是0-5，tag也有数据

LarryDong, 2024-04-02:
> 您好，我采集的数据里tag始终为0，line为0-3是正常的吗，另外我在这个仓库下[https://github.com/hku-mars/LiDAR_IMU_Init下载的mid360数据line是0-5，tag也有数据](https://github.com/hku-mars/LiDAR_IMU_Init%E4%B8%8B%E8%BD%BD%E7%9A%84mid360%E6%95%B0%E6 [...]

正常。mid360是4个扫描头，scan是4。那个数据不是mid360，好像是avia。

closed 2024-04-02 by LarryDong

danilo-schneider, 2025-06-16:
Actually it is from 0 to 4 (0, 1, 2, 3, 4) and increments from point to point.