Livox-SDK/livox_ros_driver2 #78: How to get topics from the multiple Livox Mid360 LiDARs

Blacook, 2023-09-06:
I tried to get 3Dcloud topics from two LiDARs. 
I change the multi_topic 0 to 1 in the msg_MID360.launch.
And I add another LiDAR config in MID360_config.json following the README.

As a result, I could list the topic name by the "rostopic list" command.
However, I could not echo the content of both topics by the "rostopic echo".

satyajitghana, 2023-10-17:
I would suggest you to get a Livox Hub, much easier to work with that.

Shidabot, 2024-07-17:
> I tried to get 3Dcloud topics from two LiDARs. I change the multi_topic 0 to 1 in the msg_MID360.launch. And I add another LiDAR config in MID360_config.json following the README.
>
> As a result, I [...]

Have you ever solved that？

JoLichtenfeld, 2025-01-15:
Have you figured out a solution to this?

Hotsburp, 2025-02-22:
@JoLichtenfeld and @Blacook, you can look at issue #85, it worked for me using ROS1 and Melodic on Ubuntu 18.04. Ideally, you will just need a single .json file for both lidars and each lidar should have its own .launch file (remember to turn on multiple topic). 

This solves the issue of being able to see both the lidar point cloud data at the same time and also provide unique names for the frames of each of the lidars.

KimHyung, 2025-02-24:
1.Change LiDAR IP
- To change the IP of Livox Mid360 LiDAR, you need to use Livox Viewer 2.
ex) lidar1 : 192.168.0.101, lidar2 : 192.168.0.102

2.Modify config.json file
- Change the LiDAR IPs to match the updated settings (192.168.0.101, 192.168.0.102).
- Ensure cmd_data_ip match your network environment.
ex)
{
  "lidar_summary_info" : {
    "lidar_type": 8
  },
  "MID360": {
    "lidar_net_info" : {
      "cmd_data_port": 56100,
      "push_msg_port": 56200,
      "point_data_port": 56300,
      "imu_data_port": 56400,
      "log_data_port": 56500
    },
    "host_net_info" : {
      "cmd_data_ip" : "192.168.0.10",
      "cmd_data_port": 56101,
      "push_msg_ip": "192.168.0.10",
      "push_msg_port": 56201,
      "point_data_ip": "192.168.0.10",
      "point_data_port": 56301,
      "imu_data_ip" : "192.168.0.10",
      "imu_data_port": 56401,
      "log_data_ip" : "",
      "log_data_port": 56501
    }
  },
  "lidar_configs" : [
    {
      "ip" : "192.168.0.101",
      "pcl_data_type" : 1,
      "pattern_mode" : 0,
      "extrinsic_parameter" : {
        "roll": 0.0,
        "pitch": 0.0,
        "yaw": 0.0,
        "x": 0,
        "y": 0,
        "z": 0
      }
    },
    {
      "ip" : "192.168.0.102",
      "pcl_data_type" : 1,
      "pattern_mode" : 0,
      "extrinsic_parameter" : {
        "roll": 0,
        "pitch": 0,
        "yaw": 0.0,
        "x": 0,
        "y": 0,
        "z": 0
      }
    }
  ]
}
3. Modify the launch File Parameters
- To handle multiple LiDARs properly, enable multi-topic mode in the launch file.
 ex) <arg name="multi_topic" default="1"/>

lexuantung9999, 2025-04-02:
Hi @KimHyung  i have a question. When you use 2 livox lidars with these setups: multi_topic = 0, publish_freq=10, which is the real frequency of topic /livox/lidar? 20Hz or 10Hz. I faced that when I ran 2 lidars, the hz of topic is double than my config, and I don't know it's right or not.