Livox-SDK/livox_ros_driver2 #79: error to install in clean ubuntu/noetic installation.

FPSychotic, 2023-09-09:
```
Base path: /home/nrover/livox_ws
Source space: /home/nrover/livox_ws/src
Build space: /home/nrover/livox_ws/build
Devel space: /home/nrover/livox_ws/devel
Install space: /home/nrover/livox_ws/install
Creating symlink "/home/nrover/livox_ws/src/CMakeLists.txt" pointing to "/opt/ros/noetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/nrover/livox_ws/src -DCATKIN_DEVEL_PREFIX=/home/nrover/livox_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nrover/livox_ws/install -G Unix Makefiles" in "/home/nrover/livox_ws/build"
####
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/nrover/livox_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nrover/livox_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
WARNING: package "depthai_ros_driver" should not depend on metapackage "image_transport_plugins" but on its packages instead
WARNING: package "depthai_ros_driver" should not depend on metapackage "image_pipeline" but on its packages instead
WARNING: package "depthai_filters" should not depend on metapackage "image_transport_plugins" but on its packages instead
WARNING: package "depthai_filters" should not depend on metapackage "image_pipeline" but on its packages instead
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - livox_ros_driver2
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'livox_ros_driver2'
-- ==> add_subdirectory(livox_ros_driver2-master)
-- livox_ros_driver2 version: 1.0.0
CMake Error at livox_ros_driver2-master/CMakeLists.txt:224 (find_package):
  By not providing "Findament_cmake_auto.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_auto", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_auto"
  with any of the following names:

    ament_cmake_autoConfig.cmake
    ament_cmake_auto-config.cmake

  Add the installation prefix of "ament_cmake_auto" to CMAKE_PREFIX_PATH or
  set "ament_cmake_auto_DIR" to a directory containing one of the above
  files.  If "ament_cmake_auto" provides a separate development package or
  SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/nrover/livox_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nrover/livox_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
nrover@ubuntu:~/livox_ws$ 

```
I had installed the driver several times and didn't have problem , also tried with catkin_make just it doesn't work.

ikhann, 2024-02-07:
`source /opt/ros/noetic/setup.bash`

yuanyuan-lalala, 2024-09-10:
you cannot use catkin_make in the folder called catkin_ws...
just use the same method as the tutorial in the readme....
---------------------------------------------------------------------------------------
source /opt/ros/noetic/setup.sh
./build.sh ROS1