Livox-SDK/livox_ros_driver2 #82: MID360和主机不在同一个网络的情况怎么办

245680, 2023-10-13:
如果我用的是wsl2,或者主机在雷达网络的NAT网络下
也就是说主机能看见雷达,雷达看不见主机,这种情况应该怎么办
我想通过端口映射将雷达和主机变成同一个网络,但是不知道雷达用的是哪个端口

georgflick, 2023-10-13:
I don't know if I understood your question correctly, but you can define the ports from your host and the lidar in the config:
https://github.com/Livox-SDK/livox_ros_driver2/blob/852c1475b386f107298082d0a540e304f30c6d14/config/MID360_config.json#L5 under
`lidar_net_info` and `host_net_info`

245680, 2023-10-13:
case 1:
I have a PC with an IP address of 192.168.0.2. A router with a WLAN IP address of 192.168.0.3 and a LAN IP range of 192.168.1.x . A MID-360, with an IP address of 192.168.1.12, is connected to the router. How to access the MID-360’s data from my PC.
case 2:
my PC is running Windows 10 with an IP 192.168.1.5 , and ROS is installed in WSL2 with an IP address of 172.240.0.3. MID-360 is connected through a network interface, and its IP is 192.168.1.12. How to access it from WSL2.

TheSeanParker, 2024-04-24:
I find a tech paper maybe help you
https://zhuanlan.zhihu.com/p/694345785

leeleebaba, 2024-12-16:
hello, i have the same problem, how can you solve it