Livox-SDK/livox_ros_driver2 #83: Livox MID360 Bulge Issue

satyajitghana, 2023-10-17:
Hi,

I've been experimenting with MID360, and we found that the raw point cloud from the device itself is giving us a bulge,

[image]

You can notice the floor and the roof part of the point cloud above. There is a Curve

I've also noticed that we don't see this kind of bulge on Livox Avia Point Clouds.

What i've tried.
- acquire raw point cloud from livox viewer, .lvx and obtain cloud from that, - results in bulge
- upgrade firmware to latest, - no change
- Try another new MID360 - same issue, slight improvement in point cloud noise, less noisy

What i think is that due to the spherical shape of the MID360, the point cloud correction is not done properly, or the hardware factory calibration is bad.

How can i solve this issue?

satyajitghana, 2023-10-17:
@Livox-Infra

ryanweii7, 2023-10-19:
Hi, this is livox support

Would you mind sending a bit of point cloud data for us to look at? I will explain the data I need

1.debug point cloud data. There is sample in the SDK -- debug_point_cloud. You can directly run and collect 15s data

2. Use livox viewer2 or sample -- livox_lidar_quick_start to collect point cloud data in lvx2 format. 15s of data is required

Please upload the data to the web disk address：
https://pan-sec.djicorp.com/s/F6k3pcX4EJPp4cG

ahar, 2023-11-03:
Hi,

Has it been solved?