Livox-SDK/livox_ros_driver2 #87: colcon build Error

2276306147, 2023-11-12:
CMake Error at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:240 (list):
  list index: 1 out of range (-1, 0)
Call Stack (most recent call first):
  CMakeLists.txt:241 (rosidl_generate_interfaces)

CMake Error at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:240 (list):
  list index: 1 out of range (-1, 0)
Call Stack (most recent call first):
  CMakeLists.txt:241 (rosidl_generate_interfaces)

CMake Error at /opt/ros/humble/share/rosidl_generator_c/cmake/rosidl_generator_c_generate_interfaces.cmake:69 (message):
  Target dependency
  '/ws_livox/build/livox_ros_driver2/rosidl_adapter/livox_ros_driver2/msg/CustomPoint.idl'
  does not exist
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:241 (rosidl_generate_interfaces)

But there is no shortage of msg

closed 2023-11-14 by 2276306147

Hanranyu, 2024-09-09:
could you tell me how to solve this problem? i stuck with it for more than 4 days, it's killing me

JkzJkz233, 2025-02-18:
Do not contain any Chinese character in your workshop path, try another path

a-nanasov, 2025-03-09:
> Do not contain any Chinese character in your workshop path, try another path

Thank you, I had the same issue. Unfortunately, tutorials documentation doesn't mention that ros doesn't support Unicode characters in paths.