Livox-SDK/livox_ros_driver2 #88: Lidar MID360 detects its own center as obstacle  or show a point in ROS

FPSychotic, 2023-11-13:
Hi, in ROS, the mid-360 generate a Point in its own origen, at 0.0.0. It makes itself is detected as an obstacle in Nav Stack and make an octomap over its self damaging the planner path options or even making recovering status as the point avoid find a plan. 
This is a serious issue.
I did upgrade the firmware for the .07 version to the newer one.
It will not show that point in livox viewer 2 , only with the ROS wrapper
[image]
you can see the point isolated in the picture centre.
I use in Fast-lio or Lio-sam slam packages blind option but even so they cannot remove the point , no idea way.
In the  Livox Viewer doesn't happen but also it has a option for blind the lidar in a distance, maybe would be useful have such option here too, apart of find the issue origin and solve it.

georgflick, 2023-11-14:
I found out that points, which are published in the origin are actually measurements with no return or measurements, where the distance was too small.

To filter these points out, you have basically two options:
1) Use a self filter at the origin to filter out these values
2) Adapt the livox_ros_driver2 to not publish the points with a range/radius of 0.0 (if sensor is set to spherical coordinates):
https://github.com/Livox-SDK/livox_ros_driver2/blob/852c1475b386f107298082d0a540e304f30c6d14/src/comm/pub_handler.cpp#L429C16-L429C16
```
if (radius == 0.0f) {
  continue;
}
```
or in the cartesian case, just continue if `x y` and `z` are 0.0 here:
https://github.com/Livox-SDK/livox_ros_driver2/blob/852c1475b386f107298082d0a540e304f30c6d14/src/comm/pub_handler.cpp#L393C37-L393C37

FPSychotic, 2023-11-14:
> I found out that points, which are published in the origin are actually measurements with no return or measurements, where the distance was too small.
>
> To filter these points out, you have basically [...]

Hi, thanks by the help , I'm just final user I have no idea of coding, and I dont know what coordinates it is using or exactly how or where I need write your modification. Could you please be a little bit more specific ?

FPSychotic, 2023-11-14:
I'm receiving this error

> if `x y` and `z` are 0.0

```
awPacket&)’:
/home/ibox/mid360_ws/src/livox_ros_driver2/src/comm/pub_handler.cpp:393:41: error: expected ‘(’ before ‘x’
  393 |     points_clouds_.push_back(point); if x y and z = 0.0    // FPSychosis added: if x y and z = 0.0
      |                                         ^
      |                                         (
make[2]: *** [livox_ros_driver2/CMakeFiles/livox_ros_driver2_node.dir/build.make:193: livox_ros_driver2/CMakeFiles/livox_ros_driver2_node.dir/src/comm/pub_handler.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3408: livox_ros_driver2/CMakeFiles/livox_ros_driver2_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```

georgflick, 2023-11-14:
> > I found out that points, which are published in the origin are actually measurements with no return or measurements, where the distance was too small.
> > To filter these points out, you have basically two options:
> > 
> > 1. Use a self filter at the origin to filter out these values
> > 2. Adapt the livox_ros_driver2 to not publish the points with a range/radius of 0.0 (if sensor is set to spherical coordinates):
> >    https://github.com/Livox-SDK/livox_ros_driver2/blob/852c1475b386f107298082d0a540e304f30c6d14/src/comm/pub_handler.cpp#L429C16-L429C16
> > 
> > ```
> > if (radius == 0.0f) {
> >   continue;
> > }
> > ```
> > or in the cartesian case, just continue if `x y` and `z` are 0.0 here: https://github.com/Livox-SDK/livox_ros_driver2/blob/852c1475b386f107298082d0a540e304f30c6d14/src/comm/pub_handler.cpp#L393C37-L393C37
> 
> Hi, thanks by the help , I'm just final user I have no idea of coding, and I dont know what coordinates it is using or exactly how or where I need write your modification. Could you please be a little bit more specific ? I dont know many about github use, that links mark exactly the line where I need change that?

The easiest would be to set `pcl_data_type` to `2` (spherical coordinates) for each lidar in your config file and add the following lines in the pub_handler.cpp https://github.com/Livox-SDK/livox_ros_driver2/blob/852c1475b386f107298082d0a540e304f30c6d14/src/comm/pub_handler.cpp#L429
after: Line 429: `double radius = raw[i].depth / 1000.0;` add
```
if (radius == 0.0) {
 continue;
}
``` and compile.

georgflick, 2023-11-14:
> I'm receiving this error
>
>
> ```
> awPacket&)’:
> /home/ibox/mid360_ws/src/livox_ros_driver2/src/comm/pub_handler.cpp:393:41: error: expected ‘(’ before ‘x’
>   393 |     points_clouds_.push_back(point); if [...]

Yes, `if x y and z = 0.0 ` was only pseudo code, correct would be 
`if (point.x == 0.0 && point.y == 0.0 && point.z == 0.0) continue;`
before line https://github.com/Livox-SDK/livox_ros_driver2/blob/852c1475b386f107298082d0a540e304f30c6d14/src/comm/pub_handler.cpp#L392

FPSychotic, 2023-11-14:
> if (point.x == 0.0 && point.y == 0.0 && point.z == 0.0) continue;

thank you so much by your help , it worked I think
[image]
Thanks by share knowledge

georgflick, 2023-11-15:
> > if (point.x == 0.0 && point.y == 0.0 && point.z == 0.0) continue;
> 
> thank you so much by your help , it worked I think [image] Thanks by share knowledge

Great, glad I could help you :+1: . You can mark this issue as closed, if the issue is resolved.

closed 2023-11-15 by FPSychotic