Livox-SDK/livox_ros_driver2 #93: ./build.sh humble error Ubuntu22.04

otomakat, 2023-11-17:
I did this step.

> git clone https://github.com/Livox-SDK/livox_ros_driver2.git ws_livox/src/livox_ros_driver2
> source /opt/ros/humble/setup.sh
> ./build.sh humble

But it resulted in this error.

> Starting >>> livox_ros_driver2
> --- stderr: livox_ros_driver2                               
> ** WARNING ** io features related to pcap will be disabled
> In file included from /ws_livox/src/livox_ros_driver2/src/livox_ros_driver2.cpp:35:
> /ws_livox/src/livox_ros_driver2/src/lds_lidar.h:90:34: error: ‘kMaxLidarCount’ was not declared in this scope
>    90 |   char broadcast_code_whitelist_[kMaxLidarCount][kBroadcastCodeSize];
>       |                                  ^~~~~~~~~~~~~~
> /ws_livox/src/livox_ros_driver2/src/lds_lidar.h:90:50: error: ‘kBroadcastCodeSize’ was not declared in this scope
>    90 |   char broadcast_code_whitelist_[kMaxLidarCount][kBroadcastCodeSize];
>       |                                                  ^~~~~~~~~~~~~~~~~~
> gmake[2]: *** [CMakeFiles/livox_ros_driver2.dir/build.make:76: CMakeFiles/livox_ros_driver2.dir/src/livox_ros_driver2.cpp.o] Error 1
> gmake[2]: *** Waiting for unfinished jobs....
> In file included from /ws_livox/src/livox_ros_driver2/src/lddc.cpp:38:
> /ws_livox/src/livox_ros_driver2/src/lds_lidar.h:90:34: error: ‘kMaxLidarCount’ was not declared in this scope
>    90 |   char broadcast_code_whitelist_[kMaxLidarCount][kBroadcastCodeSize];
>       |                                  ^~~~~~~~~~~~~~
> /ws_livox/src/livox_ros_driver2/src/lds_lidar.h:90:50: error: ‘kBroadcastCodeSize’ was not declared in this scope
>    90 |   char broadcast_code_whitelist_[kMaxLidarCount][kBroadcastCodeSize];
>       |                                                  ^~~~~~~~~~~~~~~~~~
> gmake[2]: *** [CMakeFiles/livox_ros_driver2.dir/build.make:90: CMakeFiles/livox_ros_driver2.dir/src/lddc.cpp.o] Error 1
> gmake[1]: *** [CMakeFiles/Makefile2:595: CMakeFiles/livox_ros_driver2.dir/all] Error 2
> gmake: *** [Makefile:146: all] Error 2
> ---
> Failed   <<< livox_ros_driver2 [29.9s, exited with code 2]
> 
> Summary: 0 packages finished [30.6s]
>   1 package failed: livox_ros_driver2
>   1 package had stderr output: livox_ros_driver2
> 

I spent a whole day not knowing how to resolve this error...

please help me!

georgflick, 2023-11-17:
Seems like you missed to compile and install the Livox-SDK2 https://github.com/Livox-SDK/Livox-SDK2 before compiling the livox_ros_driver2.

otomakat, 2023-11-20:
I removed Livox_SDK2 that I had already installed and reinstalled it, and it worked without any problems.
thank you!

closed 2023-11-20 by otomakat