Livox-SDK/livox_ros_driver2 #103: How to stop launched Livox LIDAR HAP?

Froas, 2023-12-28:
After launching:

`ros2 lauch livox_ros_driver2 msg_HAP_launch.py`

I would like to stop publishing and turn off the Livox lidar, but I cannot.

I found this code in **src/livox_ros_driver2/launch_ROS2/msg_HAP_launch.py:**

```
return LaunchDescription([
        livox_driver,
        # launch.actions.RegisterEventHandler(
        #     event_handler=launch.event_handlers.OnProcessExit(
        #         target_action=livox_driver,
        #         on_exit=[
        #             launch.actions.EmitEvent(event=launch.events.Shutdown()),
        #         ]
        #     )
        # )
    ])

```
But if I uncomment it, I get the following error message:

```
[livox_ros_driver2_node-1] successfully parse base config, counts: 1
[livox_ros_driver2_node-1] [INFO] [1703756374.717869962] [livox_lidar_publisher]: Init lds lidar success!
[livox_ros_driver2_node-1] GetFreeIndex key:livox_lidar_1677830336.
[livox_ros_driver2_node-1] set pcl data type, handle: 1677830336, data type: 1
[livox_ros_driver2_node-1] set scan pattern, handle: 1677830336, scan pattern: 0
[livox_ros_driver2_node-1] begin to change work mode to 'Normal', handle: 1677830336
[livox_ros_driver2_node-1] successfully set data type, handle: 1677830336, set_bit: 2
[livox_ros_driver2_node-1] successfully set pattern mode, handle: 1677830336, set_bit: 0
[livox_ros_driver2_node-1] successfully set lidar attitude, ip: 192.168.1.100
[livox_ros_driver2_node-1] successfully change work mode, handle: 1677830336
[livox_ros_driver2_node-1] successfully enable Livox Lidar imu, ip: 192.168.1.100
[livox_ros_driver2_node-1] [INFO] [1703756381.245650634] [livox_lidar_publisher]: livox/imu publish use imu format
[livox_ros_driver2_node-1] [INFO] [1703756381.379768986] [livox_lidar_publisher]: livox/lidar publish use livox custom format
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[livox_ros_driver2_node-1] [INFO] [1703756383.267249357] [rclcpp]: signal_handler(signum=2)
[livox_ros_driver2_node-1] Init queue, real query size:16.
[livox_ros_driver2_node-1] Lidar[0] storage queue size: 10
[livox_ros_driver2_node-1] Livox Lidar SDK Deinit completely!
[livox_ros_driver2_node-1] lddc destory!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
[ERROR] [livox_ros_driver2_node-1]: process has died [pid 653, exit code -11, cmd '/root/ros2_ws/install/livox_ros_driver2/lib/livox_ros_driver2/livox_ros_driver2_node --ros-args -r __node:=livox_lidar_publisher --params-file /tmp/launch_params_0sj06i89 --params-file /tmp/launch_params_r3vs8m64 --params-file /tmp/launch_params_i6gg6l81 --params-file /tmp/launch_params_ivd0upv8 --params-file /tmp/launch_params_oblbposa --params-file /tmp/launch_params_fzwrr677 --params-file /tmp/launch_params_e2ov4kb2 --params-file /tmp/launch_params_vx0l1yd0 --params-file /tmp/launch_params_zyb3hlty'].
```

If someone has faced the same problem, can you help me solve it?

closed 2024-01-15 by Froas

Froas, 2024-01-19:
SetLivoxLidarWorkMode(config.handle, kLivoxLidarWakeUp, nullptr, nullptr);

Murphy41, 2024-06-03:
You saved my life, bro! Much appreciated!

xiaojinghan0902, 2025-01-23:
Sorry I am still not sure with the solution. Where in msg_HAP_launch.py do you add the line "SetLivoxLidarWorkMode(config.handle, kLivoxLidarWakeUp, nullptr, nullptr);" that slove the problem?

Murphy41, 2025-01-24:
> Sorry I am still not sure with the solution. Where in msg_HAP_launch.py do you add the line "SetLivoxLidarWorkMode(config.handle, kLivoxLidarWakeUp, nullptr, nullptr);" that slove the problem?

I can't remember what happens, but I find this line in my code:

```
void StartLidar(uint32_t handle) {
    livox_status status = SetLivoxLidarWorkMode(handle, kLivoxLidarNormal, nullptr, nullptr);
    if (status != kLivoxLidarStatusSuccess) {
        std::cerr << "Failed to start LiDAR, status: " << status << std::endl;
    } else {
        std::cout << "START LIDAR" << std::endl;
    }
}

void StopLidar(uint32_t handle) {
    livox_status status = SetLivoxLidarWorkMode(handle, kLivoxLidarWakeUp, nullptr, nullptr);
    if (status != kLivoxLidarStatusSuccess) {
        std::cerr << "Failed to switch off LiDAR, status: " << status << std::endl;
    } else {
        std::cout << "STOP LIDAR" << std::endl;
    }
} 
```

It's in a cpp code and I didn't use the Livox in ROS environment in this project. I only include the SDK.