Livox-SDK/livox_ros_driver2 #111: Why MID360 topics are varying in rosbag record.

rahulsharma11, 2024-02-12:
Hi.

I am using MID360 for my application to build the SLAM.
I am using FASTLIO for that. It usually works great.

Issue:
1) I am doing data collection for more than 40 mins at single run and splitting the dump on every 30 seconds. 
Topics should be consistent like:
bagfile1:
/livox/imu -> 6000
/livox/lidar -> 300

bagfile2:
/livox/imu -> 6000
/livox/lidar -> 300

.....

But randomly i am getting:
bagfile 4:
/livox/imu -> 5900
/livox/lidar -> 112

bagfile 10:
/livox/imu -> 5920
/livox/lidar -> 210

bagfile 20:
/livox/imu -> 5898
/livox/lidar -> 63

The topics are fine in other bag files.

This is happening so random that makes my SLAM drift. 
I am not sure why it is happening. I am taking precautions of Lidar Connections as well.
Any suggestions?

Thanks.

wuh15, 2024-06-18:
I have the same problem.Do you solved it?

rahulsharma11, 2024-06-18:
No.
Not yet