Livox-SDK/livox_ros_driver2 #112: No Point Cloud data in RViz

proxi666, 2024-02-19:
Hello,
While running Livox Mid-360, although I am seeing no error on my terminal, but still I cannot visualize the PointCloud in RViz...
Please do guide by providing some solution...

Regards.
[image]
[image]

siliblue, 2024-03-19:
are u running in docker? I contacted tech support last year and they said they couldn't run it on a virtual machine

czp0204, 2024-09-13:
same problem here. Did you fix the problem ? Sorry to bother you

utkarsh867, 2025-01-02:
I briefly faced the same issue.

Make sure that if you are using a firewall (`ufw` in my case), you allow traffic from the LiDAR's IP address. For example, I used:

```
ufw allow from 192.168.0.100
```

where the IP address of my LiDAR is set to `192.168.0.100`.