Livox-SDK/livox_ros_driver2 #116: rviz didnot show any point

hao7203, 2024-04-02:
After executing the command "roslaunch livox_ros_driver2 rviz_MID360.launch", the terminal displays the error message "Storage point data failed, can not get index", and RViz shows the error "Fixed Frame [livox_frame] does not exist". I am sure that I have changed my computer's IP address and the MID360 configuration, but the issue still persists. Has anyone encountered the same problem and could provide assistance in resolving it?
These are the screens after I executed the command
[image]
[image]
[image]

Trochilus-hust, 2024-04-23:
I meet the same problem, Have you solved it？

hao7203, 2024-04-23:
Yes, I checked the computer using a channel and found that the LiDAR IP address did not match what the user manual stated. So, I changed the LiDAR IP to what I found, and it works.

Trochilus-hust, 2024-04-23:
> Yes, I checked the computer using a channel and found that the LiDAR IP address did not match what the user manual stated. So, I changed the LiDAR IP to what I found, and it works.

Thank you! I did what you said.

yugansh-nac, 2024-07-29:
I am having the same problem in my Ubuntu 20.04 LTS system using ROS-Noetic. The user manual says 192.168.1.50 and the Livox Viewer 2 also shows 192.168.1.50 as the IMU IP. Lidar IP is different though. Do you know how you resolved it and which IP did you use eventually ? 
Its a bit odd that these Lidars were working just fine a few months ago with the same software on the same PC.

czp0204, 2024-09-13:
I encountered same question here in Ubuntu 18.04 using ROS melodic. I have my IP settings finished including host IP and lidar IP (192.168.1xx+SN number). But points didn't displayed and there also a no TF  data problem. The terminal log printed all success and no warning and faults here`ll@ll-NUC11PAHi7:~/ws_livox$ roslaunch livox_ros_driver2 rviz_MID360.launch
... logging to /home/ll/.ros/log/924f0cea-71d3-11ef-8e73-88aedd01fd51/roslaunch-ll-NUC11PAHi7-11271.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ll-NUC11PAHi7:39409/

SUMMARY
========

PARAMETERS
 * /cmdline_file_path: livox_test.lvx
 * /cmdline_str: 47MDLC50020127
 * /data_src: 0
 * /enable_imu_bag: True
 * /enable_lidar_bag: True
 * /frame_id: livox_frame
 * /multi_topic: 0
 * /output_data_type: 0
 * /publish_freq: 10.0
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /user_config_path: /home/ll/ws_livox...
 * /xfer_format: 0

NODES
  /
    livox_lidar_publisher2 (livox_ros_driver2/livox_ros_driver2_node)
    livox_rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [11281]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 924f0cea-71d3-11ef-8e73-88aedd01fd51
process[rosout-1]: started with pid [11292]
started core service [/rosout]
process[livox_lidar_publisher2-2]: started with pid [11299]
process[livox_rviz-3]: started with pid [11300]
[ INFO] [1726233898.229248362]: Livox Ros Driver2 Version: 1.2.4
data source:0.
[ INFO] [1726233898.230478070]: Data Source is raw lidar.
[ INFO] [1726233898.230680773]: Config file : /home/ll/ws_livox/src/livox_ros_driver2/config/MID360_config.json
LdsLidar *GetInstance
config lidar type: 8
successfully parse base config, counts: 1
[2024-09-13 21:24:58.230] [console] [info] set master/slave sdk to master sdk by default  [parse_cfg_file.cpp] [Parse] [82]
[2024-09-13 21:24:58.230] [console] [info] Livox lidar logger disable.  [parse_cfg_file.cpp] [Parse] [126]
[2024-09-13 21:24:58.230] [console] [info] Device type:9 point cloud data and IMU data unicast is enabled.  [params_check.cpp] [CheckLidarMulticastIp] [100]
[2024-09-13 21:24:58.230] [console] [info] Data Handler Init Succ.  [data_handler.cpp] [Init] [49]
[2024-09-13 21:24:58.231] [console] [info] Init livox lidars succ.  [device_manager.cpp] [Init] [178]
GetFreeIndex key:livox_lidar_2130815168.
[ INFO] [1726233898.231071662]: Init lds lidar successfully!
Gtk-Message: 21:30:19.636: GtkDialog mapped without a transient parent. This is discouraged.
`

hehe-mist, 2024-09-25:
@czp0204 The same problems, you solve? Please.

richard98hess444, 2024-10-04:
Hi guys, not sure if this helps but here's my debug process.

I'm using ROS2 humble and Ubuntu 22.04 with MID 360.

Having the same problem with @hao7203 , and it turns out I just forget to build the package again after modifying the config file. 

I also set my computer IPv4 adress fixed, use nmap to scan lidar's IP and ping it. Everything is fine there so I know the problem might be with my package.

leeleebaba, 2024-12-17:
thank you very much !!!

leeleebaba, 2024-12-17:
Hello, have you tried using WSL to connect to Livox? When I did this, RViz didn't output anything, and even after changing the IP address, it didn't have any effect.

richard98hess444, 2024-12-17:
@leeleebaba did you set the fixed frame to `livox_frame` in Rviz? that might be the problem.
Also, you can check if the topics like `livox/lidar` and `livox/imu` are publishing, so you know its Rviz's or driver's problem.
It is easier to understand your problem more if you could provide a screenshot or other information like, which file did you launch, what command was sent?