Livox-SDK/livox_ros_driver2 #117: Init lds lidar fail

Rafaelmeiraf, 2024-04-16:
error using ros2 run livox_ros_driver2 livox.ros.driver2_node

root@ed137ed101f1:/projeto/ws_livox# ros2 run livox_ros_driver2 livox_ros_driver2_node
[INFO] [1713263149.624969984] [livox_driver_node]: Livox Ros Driver2 Version: 1.0.0
[INFO] [1713263149.625632273] [livox_driver_node]: Data Source is raw lidar.
[INFO] [1713263149.625676440] [livox_driver_node]: Config file : path_default
LdsLidar *GetInstance
parse summary info failed, can not open file: path_default
[ERROR] [1713263149.625792531] [livox_driver_node]: Init lds lidar fail!

leeleebaba, 2024-12-16:
hello, i'm fail to build file, how can you do that?
[image]

Rafaelmeiraf, 2024-12-25:
In my case the problem was the IP address in the LIDAR MID-360. You need to change the  LIDAR IP in the config file.

rajvishnu07, 2025-09-03:
I had the same issue as well while testing it with Jetson Orin NX (16GB) and Livox MID360. I fixed it by disabling the Jetson's firewall.
 I run these commands to disable it:
`sudo apt-get install ufw`
`sudo ufw disable
 sudo apt-get install ufw
 sudo ufw disable
 sudo ufw --force reset
 sudo iptables -F
 sudo iptables -X
 sudo iptables -t nat -F
 sudo iptables -t nat -X
 sudo iptables -t mangle -F
 sudo iptables -t mangle -X
 sudo iptables -P INPUT ACCEPT
 sudo iptables -P FORWARD ACCEPT
 sudo iptables -P OUTPUT ACCEPT
 sudo systemctl stop firewalld
 sudo systemctl disable firewalld`

Then finally
`sudo reboot -i`

It worked after the reboot

[image]
[image]