Livox-SDK/livox_ros_driver2 #121: Can't receiver any data after "Init lds lidar successfully"

deeplerming, 2024-04-24:
[image]
[image]
"roslaunch livox_ros_driver2 msg_MID360.launch", I can init lidar successfully but can't receiver any data, lidar can be pinged normally ,Any idea what could be wrong?

JiaoxianDu, 2024-06-06:
Same to me, but it works fine in another ubuntu laptop.

JiaoxianDu, 2024-06-06:
I got it... I set both lidar ip and network port ip to the same, should not...

YnewZ, 2024-09-25:
I have the same problem, how do you solve it?

deeplerming, 2024-09-30:
remember to turn off the firewall first.

closed 2024-09-30 by deeplerming

Divya40404, 2024-09-30:
Same issue, did anyone solved the error

Kay2333, 2025-11-30:
you need change the MID360config.json cmd_data_ip and ip.The ip is 192.168.1.1XX(XX is SN two numbers at last)and change the ip to 192.168.1.5.Last reboot. it will work