Livox-SDK/livox_ros_driver2 #125: can't find pointcloud from rviz (livox hap)

JunseoMin, 2024-05-14:
hi I'm using livox HAP and Ubuntu 20.04

after build completed, i modified config and lauched 
```
roslaunch livox_ros_driver2 rviz_HAP.launch
```
but pointcloud doesn't appear and the log was printed
```
[2024-05-14 20:05:16.755] [console] [info] Handle detection data, handle:1677830336, dev_type:10, sn:5CWD23984105SB1, cmd_port:56000  [general_command_handler.cpp] [HandleDetectionData] [319]
[2024-05-14 20:05:17.226] [console] [info]  Receive Command: Id 258 Seq 39  [hap_command_handler.cpp] [Handle] [67]
```
like screenshot below,,,

[image]

I thought the problem wasn't because of the IP..
but for solving
I set my host PC IP like
[image]

,used livox hap as default IP (192.168.1.100)
and my config file is
```
{
  "lidar_summary_info" : {
    "lidar_type": 8
  },
  "HAP": {
    "lidar_net_info" : {
      "cmd_data_port": 56000,
      "push_msg_port": 0,
      "point_data_port": 57000,
      "imu_data_port": 58000,
      "log_data_port": 59000
    },
    "host_net_info" : {
      "cmd_data_ip" : "192.168.1.50",
      "cmd_data_port": 56000,
      "push_msg_ip": "",
      "push_msg_port": 0,
      "point_data_ip": "192.168.1.50",
      "point_data_port": 57000,
      "imu_data_ip" : "192.168.1.50",
      "imu_data_port": 58000,
      "log_data_ip" : "",
      "log_data_port": 59000
    }
  },
  "lidar_configs" : [
    {
      "ip" : "192.168.1.1",
      "pcl_data_type" : 1,
      "pattern_mode" : 0,
      "extrinsic_parameter" : {
        "roll": 0.0,
        "pitch": 0.0,
        "yaw": 0.0,
        "x": 0,
        "y": 0,
        "z": 0
      }
    }
  ]
}
```
my launch file 
```
<launch>

	<!--user configure parameters for ros start-->
	<arg name="lvx_file_path" default="livox_test.lvx"/>
	<arg name="bd_list" default="100000000000000"/>
	<arg name="xfer_format" default="0"/>
	<arg name="multi_topic" default="0"/>
	<arg name="data_src" default="0"/>
	<arg name="publish_freq" default="10.0"/>
	<arg name="output_type" default="1"/>
	<arg name="rviz_enable" default="true"/>
	<arg name="rosbag_enable" default="false"/>
	<arg name="cmdline_arg" default="$(arg bd_list)"/>
	<arg name="msg_frame_id" default="livox_frame"/>
	<arg name="lidar_bag" default="false"/>
	<arg name="imu_bag" default="false"/>
	<!--user configure parameters for ros end--> 

	<param name="xfer_format" value="$(arg xfer_format)"/>
	<param name="multi_topic" value="$(arg multi_topic)"/>
	<param name="data_src" value="$(arg data_src)"/>
	<param name="publish_freq" type="double" value="$(arg publish_freq)"/>
	<param name="output_data_type" value="$(arg output_type)"/>
	<param name="cmdline_str" type="string" value="$(arg bd_list)"/>
	<param name="cmdline_file_path" type="string" value="$(arg lvx_file_path)"/>
	<param name="user_config_path" type="string" value="$(find livox_ros_driver2)/config/HAP_config.json"/>
	<param name="frame_id" type="string" value="$(arg msg_frame_id)"/>
	<param name="enable_lidar_bag" type="bool" value="$(arg lidar_bag)"/>
	<param name="enable_imu_bag" type="bool" value="$(arg imu_bag)"/>

	<node name="livox_lidar_publisher2" pkg="livox_ros_driver2"
	      type="livox_ros_driver2_node" required="true"
	      output="screen" args="$(arg cmdline_arg)"/>

	<group if="$(arg rviz_enable)">
		<node name="livox_rviz" pkg="rviz" type="rviz" respawn="true"
				args="-d $(find livox_ros_driver2)/config/display_point_cloud_ROS1.rviz"/>
    </group>

	<group if="$(arg rosbag_enable)">
    	<node pkg="rosbag" type="record" name="record" output="screen"
          		args="-a"/>
    </group>

</launch>
```

plz let me know why pointcloud doesn't appear Thanks..!

JunseoMin, 2024-05-17:
solved

closed 2024-05-17 by JunseoMin

leoxiang66, 2024-10-24:
hi, please tell me how did you solve this problem? I've encountered the same problem.