Livox-SDK/livox_ros_driver2 #128: LIVOX_LIDAR_SDK_LIBRARY not found

Elaine-yz, 2024-06-02:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
LIVOX_LIDAR_SDK_LIBRARY
linked by target "livox_ros_driver2_node" in directory /home/fast-drone/ws_livox/src/livox_ros_driver2

-- Configuring incomplete, errors occurred!
See also "/home/fast-drone/ws_livox/build/CMakeFiles/CMakeOutput.log".
See also "/home/fast-drone/ws_livox/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
应该如何解决此条错误呢

luvwinnie, 2024-06-05:
I got this error too. How can I solve it?

Elaine-yz, 2024-06-05:
sorry，i haven't found a solution yet.

---- Replied Message ----
| From | Leow Chee ***@***.***> |
| Date | 06/05/2024 15:21 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [Livox-SDK/livox_ros_driver2] LIVOX_LIDAR_SDK_LIBRARY not found (Issue #128) |

I got this error too. How can I solve it?

luvwinnie, 2024-06-05:
Oh, I found the solution by installing this repository.
https://github.com/Livox-SDK/Livox-SDK2/tree/master

Elaine-yz, 2024-06-05:
I downloaded this SDK, but the error still exists. I was thinking it might be a problem with static IP. Perhaps I could ask how your static IP is set.

---- Replied Message ----
| From | Leow Chee ***@***.***> |
| Date | 06/05/2024 17:28 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [Livox-SDK/livox_ros_driver2] LIVOX_LIDAR_SDK_LIBRARY not found (Issue #128) |

Oh, I found the solution by installing this repository.
https://github.com/Livox-SDK/Livox-SDK2/tree/master

luvwinnie, 2024-06-05:
No you need to install the SDK not only download it.
It have the installation instructions on the readme just as follows.

```
$ git clone https://github.com/Livox-SDK/Livox-SDK2.git
$ cd ./Livox-SDK2/
$ mkdir build
$ cd build
$ cmake .. && make -j
$ sudo make install
Note :
```

Elaine-yz, 2024-06-05:
yeah，I have installed it according to GitHub's steps, but the problem still persists

---- Replied Message ----
| From | Leow Chee ***@***.***> |
| Date | 06/05/2024 17:34 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [Livox-SDK/livox_ros_driver2] LIVOX_LIDAR_SDK_LIBRARY not found (Issue #128) |

No you need to install the SDK not only download it.
It have the installation instructions on the readme just as follows.

$ git clone https://github.com/Livox-SDK/Livox-SDK2.git
$ cd ./Livox-SDK2/
$ mkdir build
$ cd build
$ cmake .. && make -j
$ sudo make install

luvwinnie, 2024-06-05:
May i know whats kind of error messages is it?

Elaine-yz, 2024-06-05:
I have a full class today and can only send it to you tomorrow. Perhaps you have time tomorrow to help me check this error?

---- Replied Message ----
| From | Leow Chee ***@***.***> |
| Date | 06/05/2024 17:38 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [Livox-SDK/livox_ros_driver2] LIVOX_LIDAR_SDK_LIBRARY not found (Issue #128) |

May i know whats kind of error messages is it?

luvwinnie, 2024-06-05:
I'm not sure I can response instantly, but you can send the errors once I got time I can response to you.

Elaine-yz, 2024-06-05:
okay，thanks

---- Replied Message ----
| From | Leow Chee ***@***.***> |
| Date | 06/05/2024 17:48 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [Livox-SDK/livox_ros_driver2] LIVOX_LIDAR_SDK_LIBRARY not found (Issue #128) |

I'm not sure I can response instantly, but you can send the errors once I got time I can response to you.

Elaine-yz, 2024-06-06:
hello，I encountered an error as shown in the picture while installing the configuration file for sdk2.
[image]
[image].Today I was able to run the command "./build. sh ROS1". However, when running Livox ROS driver 2, the same error occurred with the command "roslaunch livox_ros_driver2 rviz_HAP. launch" as mentioned above.
[image]
[image].It seems that there is a problem with the link to the LiDAR. Perhaps you have a solution and idea to solve the problem

georgflick, 2024-06-06:
Do you have an HAP lidar? 
If so, you need to adjust the IP of the sensor in the [HAP_config:28](https://github.com/Livox-SDK/livox_ros_driver2/blob/b6ff7d1c8b210a96b74e919823771e1e32249758/config/HAP_config.json#L28) and change the IP's in the host_net_info to your local host ip of your computer.

> hello，I encountered an error as shown in the picture while installing the configuration file for sdk2. <img alt="屏幕截图 2024-06-06 170917" width="671" src="https://private-user-images.githubusercontent.com/136625689/337205336-9edf9858-158c-472e-83f8-8068b34565a5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc2Njc3NzYsIm5iZiI6MTcxNzY2NzQ3NiwicGF0aCI6Ii8xMzY2MjU2ODkvMzM3MjA1MzM2LTllZGY5ODU4LTE1OGMtNDcyZS04M2Y4LTgwNjhiMzQ1NjVhNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNlQwOTUxMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NDg2ZmRlZGI1OTI3NDEwMDcwZmMyYjZjZTQzNmE2OTM2OGM2YTU2MTMwNGViYzFkZTMxZjI1ZGM1NjliODBmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kIjGbhf-7uf0TONKk9LorO34I7qfTlqtS4-BMqjwBoE"> ![11](https://private-user-images.githubusercontent.com/136625689/337202418-5070c9d7-2fa8-47b5-9045-4190717bf7d8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc2Njc3NzYsIm5iZiI6MTcxNzY2NzQ3NiwicGF0aCI6Ii8xMzY2MjU2ODkvMzM3MjAyNDE4LTUwNzBjOWQ3LTJmYTgtNDdiNS05MDQ1LTQxOTA3MTdiZjdkOC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNlQwOTUxMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MmJiMTVmNzcyYzc3M2RiZDM2NTMyNDE2MWE4YjZjMzBhMzk1MTM3YTIwM2EzMTg4NjExMThkOTFhYmZmYWM3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.eB07lsKVs5zw3V6rjhgY-a0HchrjrefpgBRwr4qBj-Q).Today I was able to run the command "./build. sh ROS1". However, when running Livox ROS driver 2, the same error occurred with the command "roslaunch livox_ros_driver2 rviz_HAP. launch" as mentioned above. <img alt="屏幕截图 2024-06-06 170801" width="680" src="https://private-user-images.githubusercontent.com/136625689/337205074-0413ec1d-9852-4701-b840-e61a9514506a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc2Njc3NzYsIm5iZiI6MTcxNzY2NzQ3NiwicGF0aCI6Ii8xMzY2MjU2ODkvMzM3MjA1MDc0LTA0MTNlYzFkLTk4NTItNDcwMS1iODQwLWU2MWE5NTE0NTA2YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNlQwOTUxMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNGJjMGNiYzQ5ZjhmY2NhZjkxNjZhZmViOWRiMTExOTFmNDUzOGZiMzIxOThlZGM0N2YxMWI2OGJkMjcwOGM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RgHZSeUAzcOCyc7MH8m1gU2DfIAsUS9Znb-_RhOFNNc"> ![22](https://private-user-images.githubusercontent.com/136625689/337205142-09bcd9e4-0910-4302-a731-ac541b7ede7b.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc2Njc3NzYsIm5iZiI6MTcxNzY2NzQ3NiwicGF0aCI6Ii8xMzY2MjU2ODkvMzM3MjA1MTQyLTA5YmNkOWU0LTA5MTAtNDMwMi1hNzMxLWFjNTQxYjdlZGU3Yi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNlQwOTUxMTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YWQ2MjEwMzU3ZWM3NWQ0MGU5ZTA0NDkxYjdiMzBlZTExZDBiYWRkNTQ4NzUwZWNkYzQ4MzVhMjU1ZTZkYTBlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Jk0rsV6DFgT-PQEdLmoIIT3SWutq8f16vbjy0dMQ3X0).It seems that there is a problem with the link to the LiDAR. Perhaps you have a solution and idea to solve the problem

Elaine-yz, 2024-06-06:
thank  you  very  much~！I am using MID-360 and I have found a solution! It turned out that my Ethernet IP was set incorrectly.

CLearner123, 2024-09-04:
don`t forget "sudo make install"

marapakabannudeepak, 2025-02-19:
CMake Error at CMakeLists.txt:76 (target_link_libraries):
  Cannot specify link libraries for target "livox_ros_driver2" which is not
  built by this project.
I am getting this error