Livox-SDK/livox_ros_driver2 #131: Ubuntu 24.04/ ROS2 jazzy compatibility

cn999, 2024-06-25:
I want to use the LIVOX HAP with Ubuntu 24.04 LTS and ROS2 Jazzy. 
Thanks to https://github.com/Livox-SDK/Livox-SDK2/issues/76 I was able to build livox_sdk2 but building the livox_ros_driver2 fails.
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
myws2/src/livox_ros_driver2/LIVOX_INTERFACES_INCLUDE_DIRECTORIES

Any ideas? 
Thanks!

JoLichtenfeld, 2024-06-25:
I'm not sure about this specific error, but you might want to checkout our fork for ros2:
https://github.com/tu-darmstadt-ros-pkg/livox_ros_driver2

I am able to build it with Ubuntu 24.04.

cn999, 2024-06-26:
That worked for me aswell. Thanks!

cn999, 2024-06-26:
> I'm not sure about this specific error, but you might want to checkout our fork for ros2: https://github.com/tu-darmstadt-ros-pkg/livox_ros_driver2
> 
> I am able to build it with Ubuntu 24.04.

In case you also have any experience in simulating with ROS2 jazzy and gazebo instead of gazebo classic let me know. 
https://github.com/Livox-SDK/livox_laser_simulation/issues/28

xx-tao, 2024-11-25:
@cn999 Hi, Could you tell me how to proceed? I try to exec 'colcon build' but get errors.
  Could not find a package configuration file provided by "livox_sdk2" with
  any of the following names:

    livox_sdk2Config.cmake
    livox_sdk2-config.cmake
but i have install liblivox_lidar_sdk_shared.so  liblivox_lidar_sdk_static.a in /usr/local/lib

JoLichtenfeld, 2024-11-25:
It seems like you are using our fork @xx-tao ?
I recently turned the Livox SDK into a ROS2 package, too. Here's the fork: https://github.com/tu-darmstadt-ros-pkg/Livox-SDK2/tree/jazzy

Then I added it as a dependency to the livox_ros_driver2 pkg. This allows to install/build the whole thing with colcon and rosdep. Soyou can try to build the livox_sdk using our fork, let me know if it works. I will also add an info for this in the README.

xx-tao, 2024-11-25:
Thanks @JoLichtenfeld , This fork work for me.

greymfm, 2025-01-19:
This works for me on the original Livox ROS driver 2 package:

colcon build --cmake-args -DROS_EDITION=ROS2 -DCMAKE_BUILD_TYPE=Release -DHUMBLE_ROS=humble

JoLichtenfeld, 2025-10-23:
> no. I can't build on macos.

Never tested it on macos. But you did not specify any error?

skylinephotometrics, 2026-08-16:
Same error on Ubuntu 24.04 / ROS2 Jazzy (MID360S).

Cause: the if(... "humble" OR ... "jazzy") branch in CMakeLists.txt handles includes via target_link_libraries and never sets LIVOX_INTERFACES_INCLUDE_DIRECTORIES — it's only assigned in the else() branch. But target_include_directories(...) still references ${LIVOX_INTERFACES_INCLUDE_DIRECTORIES}, so on Jazzy it's undefined → NOTFOUND.

Fix: comment out that reference inside target_include_directories (~line 300):

cmake
#    ${LIVOX_INTERFACES_INCLUDE_DIRECTORIES}   # for custom msgs

Safe on Jazzy since the headers already come through target_link_libraries. Setting the variable to "" does not work — CMake rejects the empty argument; the line must be removed.