Livox-SDK/livox_ros_driver2 #133: Init IDs lidar failing occasionally

JoLichtenfeld, 2024-07-12:
Hi, 
about half of the times I launch the msg_MID360_launch.py everything works fine. But sometimes it failes:

> [livox_ros_driver2_node-1] successfully parse base config, counts: 2
> [livox_ros_driver2_node-1] bind failed
> [livox_ros_driver2_node-1] Failed to init livox lidar sdk.
> [livox_ros_driver2_node-1] [ERROR] [1720773038.203189433] [livox_lidar_publisher]: Init lds lidar fail!
> 
This happens on the same machine without changing any config and even within the same minute.
Eventually, after many trials, it will work again. Any explanation for this weird behaviour?

KAI-Ice, 2024-09-28:
我运行"ros2 launch livox_ros_driver2 rviz_MID360_launch.py"这条命令也是出现这个错误,有谁知道怎么解决吗?

JuanLassoVelasco, 2024-10-15:
Running on Jetson orin nano and also having this problem.

JoLichtenfeld, 2025-01-03:
Any updates on this?

LukasLange28, 2025-01-29:
I get this error when the "host ip" in the config file is not the ip of the pc that runs the package. So changing the "host ip" fixed it.

JoLichtenfeld, 2025-01-29:
The thing is that this happens only one out of ~ten times. Host ip is set correctly.

QingKe-UP, 2025-07-14:
@JoLichtenfeld 
Hello, have you found the cause of the problem or have you solved it.

JoLichtenfeld, 2025-07-14:
Unfortunately not.