Livox-SDK/livox_ros_driver2 #136: By not providing "Findament_cmake_auto.cmake" in CMAKE_MODULE_PATH this

Kamie1103, 2024-07-22:
`livox_ros_driver2 git:(master) ✗ sudo ./build.sh humble

/opt/ros/humble/setup.zsh: line 6: ${(%):-%N}: bad substitution
/opt/ros/humble/setup.zsh: line 6: cd: -q: invalid option
cd: usage: cd [-L|[-P [-e]] [-@]] [dir]
/opt/ros/humble/setup.zsh: line 22: /setup.sh: No such file or directory
Working Path: /home/seven/livox_ws/src/livox_ros_driver2
ROS version is: ROS2
Starting >>> livox_ros_driver2
Starting >>> livox_sdk2
--- stderr: livox_ros_driver2
CMake Error at CMakeLists.txt:224 (find_package):
  By not providing "Findament_cmake_auto.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_auto", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_auto"
  with any of the following names:

    ament_cmake_autoConfig.cmake
    ament_cmake_auto-config.cmake

  Add the installation prefix of "ament_cmake_auto" to CMAKE_PREFIX_PATH or
  set "ament_cmake_auto_DIR" to a directory containing one of the above
  files.  If "ament_cmake_auto" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< livox_ros_driver2 [0.45s, exited with code 1]
Aborted  <<< livox_sdk2 [0.49s]

Summary: 0 packages finished [0.66s]
  1 package failed: livox_ros_driver2
  1 package aborted: livox_sdk2
  2 packages had stderr output: livox_ros_driver2 livox_sdk2`

I had source /opt/ros/humble/setup.zsh,but it didn't work;I had tried `sudo chmod 777 -R livox_ws`,but it happened nothing.

adijams01, 2024-08-08:
getting same error, were you able to solve this?

yuanyuan-lalala, 2024-09-10:
> getting same error, were you able to solve this?

you cannot use catkin_make in the folder called catkin_ws...
just use the same method as the tutorial in the readme....
source /opt/ros/noetic/setup.sh
./build.sh ROS1

adadKKKX, 2025-03-12:
> > getting same error, were you able to solve this?
> 
> you cannot use catkin_make in the folder called catkin_ws... just use the same method as the tutorial in the readme.... source /opt/ros/noetic/setup.sh ./build.sh ROS1

are u serious??????

[image]

adijams01, 2025-03-15:
> > > getting same error, were you able to solve this?
> > 
> > 
> > you cannot use catkin_make in the folder called catkin_ws... just use the same method as the tutorial in the readme.... source /opt/ros/noetic/setup.sh ./build.sh ROS1
> 
> are u serious??????
> 
> [image]

I tried doing ./build.sh ROS2 instead of ./build.sh humble , haha didn't noticed there was different one for humble.
Thank you!