Livox-SDK/livox_ros_driver2 #139: MID-360 extrinsic parameter

kao-yang, 2024-08-08:
What is the extrinsic parameter of MID-360 between /livox/lidar and /livox/imu ? Where is the center of lidar?

TZECHIN6, 2024-08-09:
Referring to the manual, the sensor light center is 47mm above the base plate.

For the imu and the sensor light extrinsic parameter, well I do not find it. Might be require them to answer for that.

closed 2024-08-12 by kao-yang

TZECHIN6, 2024-08-12:
The position of the IMU chip in the point cloud coordinates is x=11.0 mm, y=23.29 mm, z=-44.12 mm.

according to the manual

kao-yang, 2024-08-13:
Thanks! Get it!

5uvin, 2025-04-09:
Hello @TZECHIN6 can you link the source you got IMU extrinsic from? I can't seem to find the lidar  to IMU extrinsics.

georgflick, 2025-04-11:
> Hello [@TZECHIN6](https://github.com/TZECHIN6) can you link the source you got IMU extrinsic from? I can't seem to find the lidar to IMU extrinsics.

It's in the manual:
https://terra-1-g.djicdn.com/851d20f7b9f64838a34cd02351370894/Livox/Livox_Mid-360_User_Manual_EN.pdf
Page 15: `The position of the IMU chip in the point cloud coordinates is x=11.0 mm, y=23.29 mm, z=-44.12 mm.`

5uvin, 2025-04-11:
@georgflick Thank you!!