Livox-SDK/livox_ros_driver2 #140: 请问如何配置两台Livox_HAP雷达，并在Ubuntu中显示不同的话题？

leslie-fig, 2024-08-11:
我在配置两台HAP类型的激光雷达的时候，通过在HAP_config.json中加入两个雷达的ip后，再启动rviz_HAP.launch文件。可以ping通两个雷达，但是rqt_graph里面却只有一个雷达发布话题(另外一个雷达节点存在，没有发布话题)，如下图所示。
[image]
配置文件如下图所示。
[image]

leslie-fig, 2024-08-11:
我看官网上面又一个参数 lidar_ipaddr，代表自动扫描并连接同一网络中的所有雷达。不知道这个对我上面出现的问题有没有帮助(雷达不在身边，很难进行实际的测试)

leslie-fig, 2024-08-11:
> 我看官网上面又一个参数 lidar_ipaddr，代表自动扫描并连接同一网络中的所有雷达。不知道这个对我上面出现的问题有没有帮助(雷达不在身边，很难进行实际的测试)

配置成功的话，应该使用rostopic list应该出现如下的结果。
[image]

Livox-louie, 2024-09-11:
以下是一个node连接两个MID360（在同一个网段）的配置文件，可以参考着修改您的配置文件，主要是`host_net_info`的差异。另外还需要将launch文件的multi_topic修改为1:
launch:
`<arg name="multi_topic" default="1"/>`
config:
```{
  "lidar_summary_info" : {
    "lidar_type": 8  # protocol type index，please don't revise this value
  },
    "MID360": {
        "lidar_net_info": {
            "cmd_data_port": 56100, # command port
            "push_msg_port": 56200, 
            "point_data_port": 56300,
            "imu_data_port": 56400,
            "log_data_port": 56500
        },
        "host_net_info": [
            {
                "lidar_ip": ["192.168.1.107","192.168.1.148"], # Lidar ip
                "host_ip": "192.168.1.5", # host ip
                "cmd_data_port": 56101,
                "push_msg_port": 56201,
                "point_data_port": 56301,
                "imu_data_port": 56401,
                "log_data_port": 56501
            }
        ]
    },
    "lidar_configs": [
        {
            "ip": "192.168.1.107", # ip of the LiDAR you want to config
            "pcl_data_type": 1,
            "pattern_mode": 0,
            "extrinsic_parameter": {
                "roll": 0.0,
                "pitch": 0.0,
                "yaw": 0.0,
                "x": 0,
                "y": 0,
                "z": 0
            }
        },
        {
            "ip": "192.168.1.148", # ip of the LiDAR you want to config
            "pcl_data_type": 1,
            "pattern_mode": 0,
            "extrinsic_parameter": {
                "roll": 0.0,
                "pitch": 0.0,
                "yaw": 0.0,
                "x": 0,
                "y": 0,
                "z": 0
            }
        }
    ]
}```

leslie-fig, 2024-09-11:
谢谢您，现在已经解决了

 

 <arg name="multi_topic" default="1"/&gt;

closed 2024-09-29 by Livox-louie

leslie-fig, 2024-09-29:
这个比较简单，

简强
***@***.***

 
Closed #140 as completed.

leslie-fig, 2024-09-29:
简强
***@***.***

leslie-fig, 2024-09-29:
将这个修改为1，不好意思，操作有点问题。

简强
***@***.***