Livox-SDK/livox_ros_driver2 #141: cannot change publishing frequency Mid-360 in ROS1

IacopomC, 2024-08-19:
I'm using the Livox Mid-360 and ROS1 and I'm trying to increase the point cloud frequency above 10Hz. Unfortunately, changing the parameter ```publish_freq``` in the [msg_MID360.launch](https://github.com/Livox-SDK/livox_ros_driver2/blob/master/launch_ROS1/msg_MID360.launch) launch file doesn't seem to affect the frequency at all. Is there something else I need to change?

mkarklins, 2024-09-25:
The specs state that the typical frequency is 10hz ( https://www.livoxtech.com/mid-360/specs ). Do you know for sure that it can go beyond 10?

Ives-Natsume, 2025-06-26:
> The specs state that the typical frequency is 10hz ( https://www.livoxtech.com/mid-360/specs ). Do you know for sure that it can go beyond 10?

I worked with MID360 recently, instead of using livox ros driver I wrote driver myself, and I think by changing frame integration time we could get higher freq, but points would get thinner as cost. 10Hz might be a sweetpoint