Livox-SDK/livox_ros_driver2 #145: " Livox Viewer 2 - Ubuntu - 2023-02-07 " crashes in debian and ubuntu

deebuls, 2024-09-03:
I downloaded the " Livox Viewer 2 - Ubuntu - 2023-02-07 " from [here ](https://www.livoxtech.com/downloads) .

It crashed in debian 11 book worm and Ubuntu 22.04 . 

crash report 
```
[2024.09.03-10.37.57:598][ 36]ShowFlag.Lighting = "0"
[2024.09.03-10.37.57:598][ 36]ShowFlag.PostProcessing = "0"
[2024.09.03-10.37.57:598][ 36]ShowFlag.Particles = "0"
[2024.09.03-10.37.57:598][ 36]ShowFlag.AntiAliasing = "0"
[2024.09.03-10.37.57:598][ 36]LogUMG: Display: Widget Class WBP_ViewerMain_C - Loaded Fast Template.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Failed to find symbol file, expected location:
"/home/deebuls/Documents/code/LivoxViewer2/LivoxViewer2/Binaries/Linux/LivoxViewer2.sym"
[2024.09.03-10.37.57:643][ 36]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000000

[2024.09.03-10.37.57:643][ 36]LogCore: Fatal error!

0x0000000002eb3d6c LivoxViewer2!UnknownFunction(0x2cb3d6c)
0x0000000002f9b498 LivoxViewer2!UnknownFunction(0x2d9b497)
0x0000000003f6e33d LivoxViewer2!UnknownFunction(0x3d6e33c)
0x0000000003f796b7 LivoxViewer2!UnknownFunction(0x3d796b6)
0x0000000003f701dd LivoxViewer2!UnknownFunction(0x3d701dc)
0x0000000003f6fdbb LivoxViewer2!UnknownFunction(0x3d6fdba)
0x0000000003f70c20 LivoxViewer2!UnknownFunction(0x3d70c1f)
0x0000000003f701dd LivoxViewer2!UnknownFunction(0x3d701dc)
0x0000000003f6fdbb LivoxViewer2!UnknownFunction(0x3d6fdba)
0x0000000003f70c20 LivoxViewer2!UnknownFunction(0x3d70c1f)
0x0000000003f701dd LivoxViewer2!UnknownFunction(0x3d701dc)
0x0000000003e18837 LivoxViewer2!UnknownFunction(0x3c18836)
0x0000000003f75884 LivoxViewer2!UnknownFunction(0x3d75883)
0x0000000005ce0b49 LivoxViewer2!UnknownFunction(0x5ae0b48)
0x0000000005cdf0a3 LivoxViewer2!UnknownFunction(0x5adf0a2)
0x0000000005cc382d LivoxViewer2!UnknownFunction(0x5ac382c)
0x0000000005ca7e10 LivoxViewer2!UnknownFunction(0x5aa7e0f)
0x0000000005ca7bcb LivoxViewer2!UnknownFunction(0x5aa7bca)
0x0000000005cc363c LivoxViewer2!UnknownFunction(0x5ac363b)
0x0000000005c85f54 LivoxViewer2!UnknownFunction(0x5a85f53)
0x0000000005c862fe LivoxViewer2!UnknownFunction(0x5a862fd)
0x0000000005cc363c LivoxViewer2!UnknownFunction(0x5ac363b)
0x0000000005cdeea7 LivoxViewer2!UnknownFunction(0x5adeea6)
0x0000000005cc363c LivoxViewer2!UnknownFunction(0x5ac363b)
0x0000000005cdfbce LivoxViewer2!UnknownFunction(0x5adfbcd)
0x0000000005d6cd37 LivoxViewer2!UnknownFunction(0x5b6cd36)
0x0000000003e18837 LivoxViewer2!UnknownFunction(0x3c18836)
0x0000000003f6f19a LivoxViewer2!UnknownFunction(0x3d6f199)
0x0000000003f79eb9 LivoxViewer2!UnknownFunction(0x3d79eb8)
0x0000000003f701dd LivoxViewer2!UnknownFunction(0x3d701dc)
0x0000000003f6fdbb LivoxViewer2!UnknownFunction(0x3d6fdba)
0x0000000003f70c20 LivoxViewer2!UnknownFunction(0x3d70c1f)
0x0000000003f701dd LivoxViewer2!UnknownFunction(0x3d701dc)
0x0000000003f6fdbb LivoxViewer2!UnknownFunction(0x3d6fdba)
0x0000000003f70c20 LivoxViewer2!UnknownFunction(0x3d70c1f)
0x0000000003f701dd LivoxViewer2!UnknownFunction(0x3d701dc)
0x0000000003e18837 LivoxViewer2!UnknownFunction(0x3c18836)
0x0000000003f75884 LivoxViewer2!UnknownFunction(0x3d75883)
0x00000000061ecbea LivoxViewer2!UnknownFunction(0x5fecbe9)
0x0000000006203063 LivoxViewer2!UnknownFunction(0x6003062)
0x00000000062023e7 LivoxViewer2!UnknownFunction(0x60023e6)
0x0000000006fa5526 LivoxViewer2!UnknownFunction(0x6da5525)
0x00000000066a5c23 LivoxViewer2!UnknownFunction(0x64a5c22)
0x00000000066a6273 LivoxViewer2!UnknownFunction(0x64a6272)
0x0000000006f77d39 LivoxViewer2!UnknownFunction(0x6d77d38)
0x0000000006e8a7d7 LivoxViewer2!UnknownFunction(0x6c8a7d6)
0x0000000006e86ab6 LivoxViewer2!UnknownFunction(0x6c86ab5)
0x0000000006e8811a LivoxViewer2!UnknownFunction(0x6c88119)
0x000000000667b11d LivoxViewer2!UnknownFunction(0x647b11c)
0x0000000002d8c821 LivoxViewer2!UnknownFunction(0x2b8c820)
0x0000000002d935da LivoxViewer2!UnknownFunction(0x2b935d9)
0x0000000007782cfb LivoxViewer2!UnknownFunction(0x7582cfa)
0x00007fccdcc4624a libc.so.6!UnknownFunction(0x27249)
0x00007fccdcc46305 libc.so.6!__libc_start_main(+0x84)
0x0000000002d87029 LivoxViewer2!UnknownFunction(0x2b87028)

[2024.09.03-10.37.57:655][ 36]LogExit: Executing StaticShutdownAfterError
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Malloc Size=131160 LargeMemoryPoolOffset=459112 
Malloc Size=84222 LargeMemoryPoolOffset=543358 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

```

I am interested in changing ip address of the mid-360, Is there anyway to change IP address without the viewer 2 .

CraigMS-dev, 2024-09-18:
I have this issue too, did you manage to solve this? I'm also on Ubuntu 22.04.

yogeshflux, 2024-11-18:
For LivoxViewer2 for Ubuntu v2.3.0

sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt update
sudo apt upgrade

Junble19768, 2024-12-26:
I once ran LivoxViewer on my Ubuntu22.04, everything was ok at that time. Then I installed LinuxPTP v4.4, and met the same issue. I fixed it using @yogeshflux's. But LinuxPTP v4.4 didn't work with mid360, so I tried LinuxPTP v3.1.1 offered by LivoxWiki, then I met the same issue again QwQ.
It seems like it's about some incorrect libs' version.

zccnqybjce, 2025-04-04:
I met the same issue on Ubuntu 22.04.
However, using  @yogeshflux's still not fixed.
The crash report is the same.

yumeminami, 2025-06-04:
anyone found the solution?