Livox-SDK/livox_ros_driver2 #156: CustomPoint.idl does not exit

Harkerbest, 2024-12-08:
Environment is:
- Ubuntu 22.04
- ROS2 humble

Problem is:

```
ubuntu@ubuntu:~/桌面/ws_livox/src/livox_ros_driver2$ ./build.sh humble
Working Path: /home/ubuntu/桌面/ws_livox/src/livox_ros_driver2
ROS version is: ROS2
Starting >>> livox_ros_driver2
--- stderr: livox_ros_driver2                         
CMake Error at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:240 (list):
  list index: 1 out of range (-1, 0)
Call Stack (most recent call first):
  CMakeLists.txt:241 (rosidl_generate_interfaces)

CMake Error at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:240 (list):
  list index: 1 out of range (-1, 0)
Call Stack (most recent call first):
  CMakeLists.txt:241 (rosidl_generate_interfaces)

CMake Error at /opt/ros/humble/share/rosidl_generator_c/cmake/rosidl_generator_c_generate_interfaces.cmake:69 (message):
  Target dependency
  '/ws_livox/build/livox_ros_driver2/rosidl_adapter/livox_ros_driver2/msg/CustomPoint.idl'
  does not exist
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:241 (rosidl_generate_interfaces)

---
Failed   <<< livox_ros_driver2 [2.57s, exited with code 1]

Summary: 0 packages finished [2.74s]
  1 package failed: livox_ros_driver2
  1 package had stderr output: livox_ros_driver2
ubuntu@ubuntu:~/桌面/ws_livox/src/livox_ros_driver
```

I have asked gpt and searched online, couldn't find a way to solve it

HasturSTS, 2025-01-22:
Do not contain any Chinese character in your workshop path, try another path

Harkerbest, 2025-01-22:
> Do not contain any Chinese character in your workshop path, try another path

Thank you very much

closed 2025-01-22 by Harkerbest