Livox-SDK/livox_ros_driver2 #159: How can I fetch the firmware versoin from the device ?

rjwb1, 2024-12-12:


georgflick, 2024-12-13:
you can check the firmware version using the livox viewer.

rjwb1, 2024-12-28:
@georgflick I need to be able to automatically store the serial number to a file so I can track component replacements

georgflick, 2025-01-03:
I see, in the struct DetectionData there is a char array with the serial number:
https://github.com/Livox-SDK/Livox-SDK2/blob/6a940156dd7151c3ab6a52442d86bc83613bd11b/sdk_core/comm/define.h#L135

so you could print / save it to a file after that line:
https://github.com/Livox-SDK/Livox-SDK2/blob/6a940156dd7151c3ab6a52442d86bc83613bd11b/sdk_core/device_manager.cpp#L536

georgflick, 2025-01-03:
> @georgflick I need to be able to automatically store the serial number to a file so I can track component replacements

I see, in the struct DetectionData there is a char array with the serial number:
https://github.com/Livox-SDK/Livox-SDK2/blob/6a940156dd7151c3ab6a52442d86bc83613bd11b/sdk_core/comm/define.h#L135

so you could print / save it to a file after that line:
https://github.com/Livox-SDK/Livox-SDK2/blob/6a940156dd7151c3ab6a52442d86bc83613bd11b/sdk_core/device_manager.cpp#L536

rjwb1, 2025-01-03:
@georgflick apologies, I already store the serial number, I was supposed to say the firmware version

georgflick, 2025-01-03:
> @georgflick apologies, I already store the serial number, I was supposed to say the firmware version

there is a function `QueryLivoxLidarFirmwareVer` in the sdk:
https://github.com/Livox-SDK/Livox-SDK2/blob/6a940156dd7151c3ab6a52442d86bc83613bd11b/sdk_core/command_handler/command_impl.cpp#L134