Livox-SDK/livox_ros_driver2 #168: Docker error ERROR: Cannot load message class for [livox_ros_driver2/CustomMsg]. Are your messages built?

crazy-robotics, 2025-01-23:
I build [livox_ros_driver2](https://github.com/Livox-SDK/livox_ros_driver2) in docker and on `rostopic list` i got 
```
mubashir@mubashir:~$ rostopic list 
/livox/imu
/livox/lidar
/rosout
/rosout_agg
```
on `rostopic echo /livox/imu I got the following message 
```
mubashir@mubashir:~$ rostopic echo /livox/lidar
ERROR: Cannot load message class for [livox_ros_driver2/CustomMsg]. Are your messages built?
```
I check online posts say build the workspace I have done it twice

georgflick, 2025-01-24:
are you also sourcing that workspace in your terminal?

crazy-robotics, 2025-01-24:
Yes, I source both root installation `source /opt/ros/noetic/setup.bash` and workspace `source my_workspace/devel/setup.bash`