Livox-SDK/livox_ros_driver2 #171: ros2 jazzy不能编译

beichenlee, 2025-02-07:
ros2 jazzy不能编译

georgflick, 2025-02-07:
Can you provide more information? Which error do you get? etc.

qza36, 2025-02-08:
> Can you provide more information? Which error do you get? etc.

--- stderr: livox_ros_driver2
CMake Error: File /home/arlo/livox_ws/livox_ros_driver2/package.xml does not exist.
CMake Error at /opt/ros/jazzy/share/ament_cmake_core/cmake/core/stamp.cmake:24 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/jazzy/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:73 (stamp)
  /opt/ros/jazzy/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /opt/ros/jazzy/share/ament_cmake_auto/cmake/ament_auto_find_build_dependencies.cmake:42 (ament_package_xml)
  CMakeLists.txt:225 (ament_auto_find_build_dependencies)

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/pcl/Modules/FindFLANN.cmake:45 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable FLANN_ROOT is set to:

    /usr

  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:261 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:306 (find_flann)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:570 (find_external_library)
  CMakeLists.txt:226 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

** WARNING ** io features related to pcap will be disabled
/usr/include/apr-1.0
apr-1
CMake Error at /opt/ros/jazzy/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:57 (message):
  execute_process(/home/arlo/miniconda3/bin/python3 -m rosidl_adapter
  --package-name livox_ros_driver2 --arguments-file
  /home/arlo/livox_ws/build/livox_ros_driver2/rosidl_adapter__arguments__livox_interfaces2.json
  --output-dir
  /home/arlo/livox_ws/build/livox_ros_driver2/rosidl_adapter/livox_ros_driver2
  --output-file
  /home/arlo/livox_ws/build/livox_ros_driver2/rosidl_adapter/livox_interfaces2.idls)
  returned error code 1:

  AttributeError processing template 'msg.idl.em'

  Traceback (most recent call last):

    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 48, in evaluate_template
      _interpreter = em.Interpreter(
                     ^^^^^^^^^^^^^^

  AttributeError: module 'em' has no attribute 'Interpreter'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
               ^^^^^^
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
                     ^^^^^^^^^^^^^^^
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
      return convert_msg_to_idl(
             ^^^^^^^^^^^^^^^^^^^
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()
      ^^^^^^^^^^^^^^^^^^^^^

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  /opt/ros/jazzy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:132 (rosidl_adapt_interfaces)
  CMakeLists.txt:241 (rosidl_generate_interfaces)

---
Failed   <<< livox_ros_driver2 [1min 27s, exited with code 1]

hunheyufu, 2025-02-19:
把conda文件夹挪到别的地方试试呢

hunheyufu, 2025-02-19:
> > Can you provide more information? Which error do you get? etc.
> 
> --- stderr: livox_ros_driver2 CMake Error: File /home/arlo/livox_ws/livox_ros_driver2/package.xml does not exist. CMake Error at /opt/ros/jazzy/share/ament_cmake_core/cmake/core/stamp.cmake:24 (configure_file): configure_file Problem configuring file Call Stack (most recent call first): /opt/ros/jazzy/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:73 (stamp) /opt/ros/jazzy/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml) /opt/ros/jazzy/share/ament_cmake_auto/cmake/ament_auto_find_build_dependencies.cmake:42 (ament_package_xml) CMakeLists.txt:225 (ament_auto_find_build_dependencies)
> 
> CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/pcl/Modules/FindFLANN.cmake:45 (find_package): Policy CMP0144 is not set: find_package uses upper-case _ROOT variables. Run "cmake --help-policy CMP0144" for policy details. Use the cmake_policy command to set the policy and suppress this warning.
> 
> CMake variable FLANN_ROOT is set to:
> 
> ```
> /usr
> ```
> 
> For compatibility, find_package is ignoring the variable, but code in a .cmake module might still use it. Call Stack (most recent call first): /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:261 (find_package) /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:306 (find_flann) /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:570 (find_external_library) CMakeLists.txt:226 (find_package) This warning is for project developers. Use -Wno-dev to suppress it.
> 
> ** WARNING ** io features related to pcap will be disabled /usr/include/apr-1.0 apr-1 CMake Error at /opt/ros/jazzy/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:57 (message): execute_process(/home/arlo/miniconda3/bin/python3 -m rosidl_adapter --package-name livox_ros_driver2 --arguments-file /home/arlo/livox_ws/build/livox_ros_driver2/rosidl_adapter__arguments__livox_interfaces2.json --output-dir /home/arlo/livox_ws/build/livox_ros_driver2/rosidl_adapter/livox_ros_driver2 --output-file /home/arlo/livox_ws/build/livox_ros_driver2/rosidl_adapter/livox_interfaces2.idls) returned error code 1:
> 
> AttributeError processing template 'msg.idl.em'
> 
> Traceback (most recent call last):
> 
> ```
> File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 48, in evaluate_template
>   _interpreter = em.Interpreter(
>                  ^^^^^^^^^^^^^^
> ```
> 
> AttributeError: module 'em' has no attribute 'Interpreter'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
> 
> ```
> File "<frozen runpy>", line 198, in _run_module_as_main
> File "<frozen runpy>", line 88, in _run_code
> File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
>   sys.exit(main())
>            ^^^^^^
> File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/main.py", line 53, in main
>   abs_idl_file = convert_to_idl(
>                  ^^^^^^^^^^^^^^^
> File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
>   return convert_msg_to_idl(
>          ^^^^^^^^^^^^^^^^^^^
> File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
>   expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
> File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
>   content = evaluate_template(template_name, data)
>             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> File "/opt/ros/jazzy/lib/python3.12/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
>   _interpreter.shutdown()
>   ^^^^^^^^^^^^^^^^^^^^^
> ```
> 
> AttributeError: 'NoneType' object has no attribute 'shutdown'
> 
> Call Stack (most recent call first): /opt/ros/jazzy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:132 (rosidl_adapt_interfaces) CMakeLists.txt:241 (rosidl_generate_interfaces)
> 
> Failed <<< livox_ros_driver2 [1min 27s, exited with code 1]

把conda文件夹挪到别的地方试试呢

atinfinity, 2025-05-17:
I'm using <https://github.com/Livox-SDK/livox_ros_driver2/pull/180> as a reference. It works on ROS 2 Jazzy.