Livox-SDK/livox_ros_driver2 #177: 雷达帧率变成2000HZ

superbigzz, 2025-03-04:
在ros1中，使用livox_ros_driver2解析mid360数据，偶尔会出现发布的雷达话题帧率为2000hz，单帧点云数据很少，请问这是什么问题？
[image]

[image]

channing-prince, 2025-03-15:
遇到了同样问题，尝试了换电源，换另一个雷达，换连接线，修改发布速率等等都没解决这个问题，不知道是什么触发的，我现在是稳定出现这个情况：在雷达上电大概75s左右，根据rostopic echo /livox/lidar 反馈的数据，雷达会突然发布速率变为0，大概6 - 8s之后，发布速率变为2000hz以上，但在Livox Viewer 2 - Ubuntu中未观测到这个现象。
        现在临时解决办法就是让livox_ros_driver2 晚于雷达上电85s左右，就可以正常工作，简单来说就是跳过那一段异常就不会再出现这个情况。PS：后面有时间来补上日志数据

    I've encountered the same issue. I tried switching the power supply, using a different radar unit, replacing the cable, and adjusting the publishing rate, but none of these resolved the problem. I'm not sure what triggers it, but here's the consistent behavior I'm observing: About 75 seconds after powering on the radar, according to the data from rostopic echo /livox/lidar, the radar suddenly stops publishing data (the publishing rate drops to 0). Then, roughly 6–8 seconds later, the publishing rate jumps to above 2000 Hz. However, this phenomenon is not observed in Livox Viewer 2 on Ubuntu.

    The temporary solution I've found is to start livox_ros_driver2 about 85 seconds after powering on the radar, which allows it to work normally. In short, by skipping that abnormal period, the issue no longer occurs.

PS: I’ll add the log data later when I'm available.

FelixCooper1026, 2025-03-18:
> 遇到了同样问题，尝试了换电源，换另一个雷达，换连接线，修改发布速率等等都没解决这个问题，不知道是什么触发的，我现在是稳定出现这个情况：在雷达上电大概75s左右，根据rostopic echo /livox/lidar 反馈的数据，雷达会突然发布速率变为0，大概6 - 8s之后，发布速率变为2000hz以上，但在Livox Viewer 2 - Ubuntu中未观测到这个现象。 [...]

**点云发布频率是由 pub_handler.cpp 中的 PubHandler::CheckTimer 函数控制的：**
https://github.com/Livox-SDK/livox_ros_driver2/blob/6b9356cadf77084619ba406e6a0eb41163b08039/src/comm/pub_handler.cpp#L195

在雷达未做时间同步的情况下，判断发布的逻辑如下：

1. 获取当前的高精度时间：
```
auto now_time = std::chrono::high_resolution_clock::now();
```
2. 检查是否该发布数据：
```
if (now_time - last_pub_time_ < std::chrono::nanoseconds(publish_interval_)) {
    return;
}
```

- `publish_interval_` 是发布间隔时间（默认10hz，每帧100ms）。

- `now_time - last_pub_time_ `计算的是当前时间和上次发布时间的差值，如果这个差值小于发布时间间隔，就直接返回，不发布数据。

简单来说： "如果距离上次发布还没到 100ms，就不做任何事，等时间到了再发布。"

3. 更新时间并发布点云数据
```
last_pub_time_ += std::chrono::nanoseconds(publish_interval_);
```

- 这里是把上次发布时间往后推 100ms，相当于“设定好下次数据发布的时间点”。

- 这样就可以确保数据是按照固定间隔发布的。

简单来说： "时间到了，发布数据，同时更新一下下一次该发布的时间。"

这里就会有一个bug，如果系统时间出现跳变，发布的频率会受到影响，具体情况取决于时间跳变的方向（向前跳跃或向后跳跃）：

**当系统时间发生向前跳变时（时间大幅增加，比如从 10s 跳到 20s）**

1. `now_time` 变得比 `last_pub_time_` 大很多。
2. `now_time - last_pub_time_` 远远大于 `publish_interval_`，满足 `if` 语句的条件。
3. 代码执行 `last_pub_time_ += std::chrono::nanoseconds(publish_interval_);`，但 `now_time` 仍然比 `last_pub_time_` 大很多。
4. 循环进入“补偿式”数据发布阶段，每次循环 `last_pub_time_` 仅增加 `publish_interval_`，但 `now_time` 依然远远领先，导致系统连续发布数据，直到 `last_pub_time_` 追上 `now_time`。

- Mid-360 的点频是 20万点/秒，正常 100ms 发布一帧时，每帧有 20000 个点，帧率为 200000/20000=10 hz；

- 当系统时间向前跳变时，由于每个 UDP 包中仅包含 96 个点（参考 Mid-360 数据协议），基于上面的发布逻辑，每个 UDP 包都被当做完整的一帧发布，这时帧率为 200000 / 96 ≈ 2083 hz

**系统时间向后跳变（时间大幅减少，比如从 20s 跳到 10s）**

1. `now_time` 变小：
2. `now_time - last_pub_time_` 可能变为负值或一个很小的数值。
3. 代码会认为数据发布的间隔尚未达到 `publish_interval_`，导致数据发布暂停，直到系统时间追上 `last_pub_time_`。

回到@superbigzz @channing-prince 的问题，可以检查一下出现该问题时系统时间有没有跳变 / 雷达上电75s~85s左右系统时间有没有跳变。

### 如何修复时间跳变的问题？

1. 为了防止系统时间跳变影响数据发布，可以改用 基于 `steady_clock` 的时间源，例如：
```
auto now_time = std::chrono::steady_clock::now();
```

`steady_clock` 不会受系统时间跳变影响，可以保证时间间隔的稳定性。这样，无论系统时间如何调整，发布数据的时间间隔都不会受到影响。

2. 如果必须使用 `high_resolution_clock`（例如需要与系统时间同步），可以在计算时间间隔时做额外的判断，例如：

```
if (now_time < last_pub_time_) {
    // 说明系统时间向后跳变，重置 last_pub_time_
    last_pub_time_ = now_time;
} else if (now_time - last_pub_time_ >= std::chrono::nanoseconds(publish_interval_)) {
    // 说明系统时间向前跳变，重置 last_pub_time_
    last_pub_time_ = now_time;
} 
```

这样可以确保 `last_pub_time_` 不会超前或滞后于 `now_time`，从而避免时间跳变引起的数据发布频率异常问题。

channing-prince, 2025-04-01:
前情提要：

在办公室测试算法Fast-Lio时发现，定位数据会在系统启动80s左右乱掉，xyz均跳变到相当大的值：

> 雷达之前情况是输出速率突然到0，接着过几秒变成200。现在启用了ptpd时间同步，雷达输出速率会一直保持在0。

```
average rate: 40.027
	min: 0.016s max: 0.028s std dev: 0.00078s window: 561
average rate: 40.025
	min: 0.016s max: 0.028s std dev: 0.00077s window: 601
average rate: 40.023
	min: 0.016s max: 0.028s std dev: 0.00079s window: 641
average rate: 40.022
	min: 0.016s max: 0.028s std dev: 0.00079s window: 681
average rate: 40.021
	min: 0.016s max: 0.028s std dev: 0.00078s window: 721
average rate: 40.019
	min: 0.016s max: 0.029s std dev: 0.00083s window: 758
cat@lubancat:~$ rostopic hz /livox/lidar 
subscribed to [/livox/lidar]
average rate: 0.000
	min: 0.024s max: 24824488.734s std dev: 3973722.88819s window: 39
average rate: 0.000
	min: 0.022s max: 24824488.734s std dev: 2775239.84689s window: 80
average rate: 0.000
	min: 0.018s max: 24824488.734s std dev: 2266075.39201s window: 120
average rate: 0.000
	min: 0.018s max: 24824488.734s std dev: 1962509.33664s window: 160
average rate: 0.000
	min: 0.009s max: 24824488.734s std dev: 1755334.26737s window: 200
average rate: 0.000
	min: 0.009s max: 24824488.734s std dev: 1602399.82923s window: 240
average rate: 0.000

```

雷达定位输出

```
---
header: 
  seq: 3310
  stamp: 
    secs: 1743480796
    nsecs:  50247431
  frame_id: "odom"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: -221586299.5883662
      y: -138713664.6661016
      z: -427235201.94158465
    orientation: 
      x: -0.8589724572817269
      y: 0.3953147206669799
      z: 0.017781754649646764
      w: -0.32492521979208616
  covariance: [4.1178228647096676e-08, -2.261366523707484e-08, -3.0090717293372595e-07, -9.342576484202558e-06, -8.36902533911829e-06, 1.0613216031724697e-05, -2.2619992387386095e-08, 1.242209564995578e-08, 1.652940110131809e-07, 5.1354141979192465e-06, 4.598628473845447e-06, -5.832873708085977e-06, -3.010065202234701e-07, 1.6530079368229676e-07, 2.1995694310106708e-06, 6.833647683429232e-05, 6.122306681466188e-05, -7.763526712160934e-05, -9.45786632479333e-06, 5.092859332139868e-06, 6.83237221836806e-05, 0.003281749907079849, 0.002708209151354302, -0.0035857082682709436, -8.450124420557775e-06, 4.587661256181717e-06, 6.121817756132657e-05, 0.0027079332848402426, 0.0030832319379395685, -0.003464597316542327, 1.0682895575586323e-05, -5.840305927155772e-06, -7.762833312055761e-05, -0.003586643854464132, -0.0034647056690389304, 0.004238802300162728]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 3311
  stamp: 
    secs: 1743480796
    nsecs:  75207472
  frame_id: "odom"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: -221723042.0290252
      y: -138864547.53921312
      z: -427475209.24559253
    orientation: 
      x: -0.8570847610958776
      y: 0.4007070609915473
      z: 0.01202298051457411
      w: -0.32356608522566377
  covariance: [3.53970350834472e-08, -2.0824850999909204e-08, -2.500569459898166e-07, -2.907331962540936e-06, -3.949185903357028e-06, 4.117323969511355e-06, -2.0783949995508763e-08, 1.2227113853391684e-08, 1.468251800699006e-07, 1.7152564491105567e-06, 2.324355714861836e-06, -2.4257556087046794e-06, -2.503152104503965e-07, 1.4725866273755772e-07, 1.7683075020871368e-06, 2.067148423875797e-05, 2.8035877454751638e-05, -2.9248328519418864e-05, -2.9636498152580315e-06, 1.5926882095296962e-06, 2.066837033150911e-05, 0.0029822159719932115, 0.0025483585141368795, -0.003308986202856211, -3.9735411850813515e-06, 2.232910382252997e-06, 2.8040120712823834e-05, 0.002547257313739445, 0.0031459102437766326, -0.0034037091060731862, 4.25139292962159e-06, -2.3165137316860125e-06, -2.923126759738897e-05, -0.0033089471897768605, -0.0034045262375281063, 0.004033793663558681]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---

```

之前这位大佬@FelixCooper1026讲明了原因，实在是帮大忙了，知道了原因在于电脑的时间跳变。

由于对驱动代码不熟悉，不知道`steady_clock`与`high_resolution_clock`的区别在哪，并且由于算法数据源有两个，一个是雷达，一个是飞控IMU，需要严格同步时间，便没有擅自改动源码。想到如果是时间跳变那同步一下就好，启用了ptpd同步电脑与雷达时间

```bash
sudo ptpd -M -i eth0 -C
```

但仍旧会有问题，雷达话题发布速率还是会在80s左右突然降到0

后面突然意识到，我的机载电脑关机就完全断电，重新开机时间会重置，观察发现，正好是80s左右，电脑会在有网的情况下完成时间的校正。

再次测试，在不联网环境，或者联网，但在启动驱动前校准好电脑时间，均无问题

感谢各位的帮助！

FlinkTheSecond, 2025-07-31:
> > 遇到了同样问题，尝试了换电源，换另一个雷达，换连接线，修改发布速率等等都没解决这个问题，不知道是什么触发的，我现在是稳定出现这个情况：在雷达上电大概75s左右，根据rostopic echo /livox/lidar 反馈的数据，雷达会突然发布速率变为0，大概6 - 8s之后，发布速率变为2000hz以上，但在Livox Viewer 2 - Ubuntu中未观测到这个现象。 现在临时解决办法就是让livox_ros_driver2 晚于雷达上电85s左右，就可以正常工作，简单来说就是跳过那一段异常就不会再出现这个情况。PS：后面有时间来补上日志数据
> > I've encountered the same issue. I tried switching the power supply, using a different radar unit, replacing the cable, and adjusting the publishing rate, but none of these resolved the problem. I'm not sure what triggers it, but here's the consistent behavior I'm observing: About 75 seconds after powering on the radar, according to the data from rostopic echo /livox/lidar, the radar suddenly stops publishing data (the publishing rate drops to 0). Then, roughly 6–8 seconds later, the publishing rate jumps to above 2000 Hz. However, this phenomenon is not observed in Livox Viewer 2 on Ubuntu.
> > The temporary solution I've found is to start livox_ros_driver2 about 85 seconds after powering on the radar, which allows it to work normally. In short, by skipping that abnormal period, the issue no longer occurs.
> > PS: I’ll add the log data later when I'm available.
> 
> **点云发布频率是由 pub_handler.cpp 中的 PubHandler::CheckTimer 函数控制的：**
> 
> [livox_ros_driver2/src/comm/pub_handler.cpp](https://github.com/Livox-SDK/livox_ros_driver2/blob/6b9356cadf77084619ba406e6a0eb41163b08039/src/comm/pub_handler.cpp#L195)
> 
> Line 195 in [6b9356c](/Livox-SDK/livox_ros_driver2/commit/6b9356cadf77084619ba406e6a0eb41163b08039)
>  } else { // Disable time synchronization 
> 
> 在雷达未做时间同步的情况下，判断发布的逻辑如下：
> 
>     1. 获取当前的高精度时间：
> 
> 
> ```
> auto now_time = std::chrono::high_resolution_clock::now();
> ```
> 
>     2. 检查是否该发布数据：
> 
> 
> ```
> if (now_time - last_pub_time_ < std::chrono::nanoseconds(publish_interval_)) {
>     return;
> }
> ```
> 
>     * `publish_interval_` 是发布间隔时间（默认10hz，每帧100ms）。
> 
>     * `now_time - last_pub_time_ `计算的是当前时间和上次发布时间的差值，如果这个差值小于发布时间间隔，就直接返回，不发布数据。
> 
> 
> 简单来说： "如果距离上次发布还没到 100ms，就不做任何事，等时间到了再发布。"
> 
>     3. 更新时间并发布点云数据
> 
> 
> ```
> last_pub_time_ += std::chrono::nanoseconds(publish_interval_);
> ```
> 
>     * 这里是把上次发布时间往后推 100ms，相当于“设定好下次数据发布的时间点”。
> 
>     * 这样就可以确保数据是按照固定间隔发布的。
> 
> 
> 简单来说： "时间到了，发布数据，同时更新一下下一次该发布的时间。"
> 
> 这里就会有一个bug，如果系统时间出现跳变，发布的频率会受到影响，具体情况取决于时间跳变的方向（向前跳跃或向后跳跃）：
> 
> **当系统时间发生向前跳变时（时间大幅增加，比如从 10s 跳到 20s）**
> 
>     1. `now_time` 变得比 `last_pub_time_` 大很多。
> 
>     2. `now_time - last_pub_time_` 远远大于 `publish_interval_`，满足 `if` 语句的条件。
> 
>     3. 代码执行 `last_pub_time_ += std::chrono::nanoseconds(publish_interval_);`，但 `now_time` 仍然比 `last_pub_time_` 大很多。
> 
>     4. 循环进入“补偿式”数据发布阶段，每次循环 `last_pub_time_` 仅增加 `publish_interval_`，但 `now_time` 依然远远领先，导致系统连续发布数据，直到 `last_pub_time_` 追上 `now_time`。
> 
> 
>     * Mid-360 的点频是 20万点/秒，正常 100ms 发布一帧时，每帧有 20000 个点，帧率为 200000/20000=10 hz；
> 
>     * 当系统时间向前跳变时，由于每个 UDP 包中仅包含 96 个点（参考 Mid-360 数据协议），基于上面的发布逻辑，每个 UDP 包都被当做完整的一帧发布，这时帧率为 200000 / 96 ≈ 2083 hz
> 
> 
> **系统时间向后跳变（时间大幅减少，比如从 20s 跳到 10s）**
> 
>     1. `now_time` 变小：
> 
>     2. `now_time - last_pub_time_` 可能变为负值或一个很小的数值。
> 
>     3. 代码会认为数据发布的间隔尚未达到 `publish_interval_`，导致数据发布暂停，直到系统时间追上 `last_pub_time_`。
> 
> 
> 回到[@superbigzz](https://github.com/superbigzz) [@channing-prince](https://github.com/channing-prince) 的问题，可以检查一下出现该问题时系统时间有没有跳变 / 雷达上电75s~85s左右系统时间有没有跳变。
> ### 如何修复时间跳变的问题？
> 
>     1. 为了防止系统时间跳变影响数据发布，可以改用 基于 `steady_clock` 的时间源，例如：
> 
> 
> ```
> auto now_time = std::chrono::steady_clock::now();
> ```
> 
> `steady_clock` 不会受系统时间跳变影响，可以保证时间间隔的稳定性。这样，无论系统时间如何调整，发布数据的时间间隔都不会受到影响。
> 
>     2. 如果必须使用 `high_resolution_clock`（例如需要与系统时间同步），可以在计算时间间隔时做额外的判断，例如：
> 
> 
> ```
> if (now_time < last_pub_time_) {
>     // 说明系统时间向后跳变，重置 last_pub_time_
>     last_pub_time_ = now_time;
> } else if (now_time - last_pub_time_ >= std::chrono::nanoseconds(publish_interval_)) {
>     // 说明系统时间向前跳变，重置 last_pub_time_
>     last_pub_time_ = now_time;
> } 
> ```
> 
> 这样可以确保 `last_pub_time_` 不会超前或滞后于 `now_time`，从而避免时间跳变引起的数据发布频率异常问题。

我尝试过将now_time修改成steady_clcok类型变量，但在运行程序后仍然出现问题，我将系统时间向后调整（1970到2000）时，点云频率会降到0。这之后我又尝试过使用CLOCK_MONOTONIC_RAW时间，但是同样存在这个问题。
而对于原本的程序，我尝试用一下函数进行调整
    if (now_time < last_pub_time_) {
      // 说明系统时间向后跳变，重置 last_pub_time_
      std::cout << "Time Skip Backward !!! Reset Last_pub_time_ !!!" <<std::endl;
      last_pub_time_ = now_time;
      return;
    } else if (now_time - last_pub_time_ > 4 * (std::chrono::nanoseconds(publish_interval_))) {
      // 说明系统时间向前跳变
      std::cout << "Time Skip Forward !!! Reset Last_pub_time_ !!!" <<std::endl;
      last_pub_time_ = now_time;
      return;
    }
同样不起效果，点云频率会在调整系统时间后降到0，请问这可能的原因是什么？除去点云发布这边，还有其他地方与时间相关吗

JINGHANW, 2026-08-20:
[image]
类似问题， 因为做了自启服务，看到了启动时间是不对，改成了等待chronyd之后。还在测试。